dotintent/react-native-ble-plx #1203: How can i solve "BleError: Location services are disabled" error?

avcismail, 2024-05-24:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [ ] I'm running the latest version

### Question

Hi,
I want to make an application that connects to a BLE device, but I get the following error. I gave location permission to the application on my Android phone, but I still get the same error. Below I share with you the codes and the article I got support from while developing this application. I would be very happy if you could help. Thanks.

```
import { StatusBar } from "expo-status-bar";
import React, { useEffect, useState } from "react";
import { StyleSheet, Text, View, PermissionsAndroid, Platform } from "react-native";
import { BleManager, State } from "react-native-ble-plx";
import { AppRegistry } from "react-native";
import { name as appName } from "./app.json";

const manager = new BleManager();

const App = () => {
  const [isLocationOn, setIsLocationOn] = useState(false);

  async function requestAccessLocationPermission() {
    try {
      const granted = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS. ACCESS_FINE_LOCATION,
        {
          title: 'Application wants location Permission',
          message:
            'Application wants location Permission',
          buttonNeutral: 'Ask Me Later',
          buttonNegative: 'Cancel',
          buttonPositive: 'OK',
        },
      );
      if (granted === PermissionsAndroid.RESULTS.GRANTED) {
      } else {
      }
    } catch (err) {
      console.warn(err);
    }
  }

  useEffect(() => {
    const stateChangeListener = manager.onStateChange((state) => {
      console.log("onStateChange : ", state);
      if (state === State.PoweredOn) {
        scan();
      }
    }, true);

    return () => {
      stateChangeListener?.remove();
    };
  }, [manager]);

  useEffect(() => {
    let locationSubscription = null;

    const addLocationListener = async () => {
      try {
        const subscription = await SystemSetting.addLocationListener((data) => {
          console.log("Location: ", data);

          setIsLocationOn(data);
        });

        locationSubscription = subscription;
      } catch (error) {
        console.error("Error adding location listener:", error);
      }
    };
    addLocationListener();

    const stopLocationListener = () => {
      if (locationSubscription) {
        console.log("Listener stoped");

        locationSubscription.remove();

        locationSubscription = null;
      }
    };

    return () => {
      stopLocationListener();
    };
  }, [isLocationOn]);

  function scan() {
    manager.startDeviceScan(null, null, (error, device) => {
      if (error) {
        // Handle error (scanning will be stopped automatically)
        console.log(error);
        return;
      }

      // Check if it is a device, you are looking for based on advertisement data
      // or other criteria.
      if (device.name === "UART Service") {
        // Stop scanning as it's not necessary if you are scanning for one device.
        console.log("Debug 1");
        connect();
        manager.stopDeviceScan();

        // Proceed with connection.
      }
      console.log("Debug 2");
    });
  }

  const connect = async () => {
    try {
      await manager
        .connectToDevice(deviceId)
        .then((device) => {
          console.log("Connected to device:", device.name);

          // Add your logic for handling the connected device

          return device.discoverAllServicesAndCharacteristics();
        })
        .catch((error) => {
          // Handle errors
        });
    } catch (error) {
      console.error("Error connecting to device:", error);
    }
  };

  return (
    <View style={styles.container}>
      <Text>Open up App.js to start working on your app!</Text>
      <StatusBar style="auto" />
    </View>
  );
};

AppRegistry.registerComponent(appName, () => App);

const styles = StyleSheet.create({
  container: {
    flex: 1,
    backgroundColor: "#fff",
    alignItems: "center",
    justifyContent: "center",
  },
});

export default App;
```

[Article is here](https://medium.com/@akashpoovaragavan/connect-ble-devices-with-react-native-ble-plx-89f0dc907d8a)

dominik-czupryna-withintent, 2024-05-26:
You are probably using Android phone with API 31+ and I can't find permissions request for it. Are you using neverForLocation flag?
Check "Ask for permissions" section in documentation - https://dotintent.github.io/react-native-ble-plx/
or try
```js
  requestBluetoothPermission = async () => {
    if (Platform.OS === 'ios') {
      return true
    }
    if (Platform.OS === 'android') {
      const apiLevel = parseInt(Platform.Version.toString(), 10)

      if (apiLevel < 31 && PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION) {
        const granted = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION)
        return granted === PermissionsAndroid.RESULTS.GRANTED
      }
      if (PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN && PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT) {
        const result = await PermissionsAndroid.requestMultiple([
          PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN,
          PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT
        ])

        return (
          result['android.permission.BLUETOOTH_CONNECT'] === PermissionsAndroid.RESULTS.GRANTED &&
          result['android.permission.BLUETOOTH_SCAN'] === PermissionsAndroid.RESULTS.GRANTED
        )
      }
    }

    this.showErrorToast('Permission have not been granted')

    return false
  }
```

avcismail, 2024-05-27:
> You are probably using Android phone with API 31+ and I can't find permissions request for it. Are you using neverForLocation flag? Check "Ask for permissions" section in documentation - [...]

Hi, thank your for your answer. I dont use API 31+ version but i am solved this error. Wasn't open location on my device :D

intent-kacper-cyranowski, 2024-06-14:
Closing an issue as it seems to be solved

closed 2024-06-14 by intent-kacper-cyranowski