dotintent/react-native-ble-plx #1204: I'm getting an error when sending commands with WriteQuestion

avcismail, 2024-05-28:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [ ] I'm running the latest version

### Question

Hi, I get the following error when sending a command with Write. What could be causing this? My device supports BLE. 

Error: 
`Write success:  {"_manager": {"_activePromises": {}, "_activeSubscriptions": {}, "_errorCodesToMessagesMapping": {"0": "Unknown error occurred. This is probably a bug! Check reason property.", "1": "BleManager was destroyed", "100": "BluetoothLE is unsupported on this device", "101": "Device is not authorized to use BluetoothLE", "102": "BluetoothLE is powered off", "103": "BluetoothLE is in unknown state", "104": "BluetoothLE is resetting", "105": "Bluetooth state change failed", "2": "Operation was cancelled", "200": "Device {deviceID} connection failed", "201": "Device {deviceID} was disconnected", "202": "RSSI read failed for device {deviceID}", "203": "Device {deviceID} is already connected", "204": "Device {deviceID} not found", "205": "Device {deviceID} is not connected", "206": "Device {deviceID} could not change MTU size", "3": "Operation timed out", "300": "Services discovery failed for device {deviceID}", "301": "Included services discovery failed for device {deviceID} and service: {serviceUUID}", "302": "Service {serviceUUID} for device {deviceID} not found", "303": "Services not discovered for device {deviceID}", "4": "Operation was rejected", "400": "Characteristic discovery failed for device {deviceID} and service {serviceUUID}", "401": "Characteristic {characteristicUUID} write failed for device {deviceID} and service {serviceUUID}", "402": "Characteristic {characteristicUUID} read failed for device {deviceID} and service {serviceUUID}", "403": "Characteristic {characteristicUUID} notify change failed for device {deviceID} and service {serviceUUID}", "404": "Characteristic {characteristicUUID} not found", "405": "Characteristics not discovered for device {deviceID} and service {serviceUUID}", "406": "Cannot write to characteristic {characteristicUUID} with invalid data format: {internalMessage}", "5": "Invalid UUIDs or IDs were passed: {internalMessage}", "500": "Descriptor {descriptorUUID} discovery failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "501": "Descriptor {descriptorUUID} write failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "502": "Descriptor {descriptorUUID} read failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "503": "Descriptor {descriptorUUID} not found", "504": "Descriptors not discovered for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "505": "Cannot write to descriptor {descriptorUUID} with invalid data format: {internalMessage}", "506": "Cannot write to descriptor {descriptorUUID}. It's not allowed by iOS and therefore forbidden on Android as well.", "600": "Cannot start scanning operation", "601": "Location services are disabled"}, "_eventEmitter": {"_nativeModule": [Object]}, "_scanEventSubscription": null, "_uniqueId": 11}, "deviceID": "98:F4:AB:6A:FF:7A", "id": 11, "isIndicatable": false, "isNotifiable": false, "isNotifying": false, "isReadable": false, "isWritableWithResponse": true, "isWritableWithoutResponse": false, "serviceID": 8, "serviceUUID": "6e400001-b5a3-f393-e0a9-e50e24dcca9e", "uuid": "6e400002-b5a3-f393-e0a9-e50e24dcca9e", "value": "MTQ1M3Jlc2V0"}`

intent-kacper-cyranowski, 2024-06-06:
Hi, the log you provided is no error. The `console.log` is placed in `then`  method so everything should be fine. I think the logged value might be confusing as it contains `_manager` object. If you ignore it, your value looks like that:

```json
{
  "deviceID": "98:F4:AB:6A:FF:7A",
  "id": 11,
  "isIndicatable": false,
  "isNotifiable": false,
  "isNotifying": false,
  "isReadable": false,
  "isWritableWithResponse": true,
  "isWritableWithoutResponse": false,
  "serviceID": 8,
  "serviceUUID": "6e400001-b5a3-f393-e0a9-e50e24dcca9e",
  "uuid": "6e400002-b5a3-f393-e0a9-e50e24dcca9e",
  "value": "MTQ1M3Jlc2V0"
}
```

If it’s not an expected behaviour please provide more details on what is the problem.

intent-kacper-cyranowski, 2024-06-14:
Hi, did you manage to solve your issue?

intent-kacper-cyranowski, 2024-07-01:
Closing due to inactivity

closed 2024-07-01 by intent-kacper-cyranowski