dotintent/react-native-ble-plx #1205: unable to read property list from file: /Users/testuser/Desktop/bluetooth-ble/ios/bluetoothble/Info.plist: The operation couldn’t be completed. (XCBUtil.PropertyListConversionError error 2.)

avcismail, 2024-05-28:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [ ] I'm running the latest version

### Question

Hi, I get the following error when debugging on iOS. I researched this error on the internet but could not find a solution. Does anyone have any ideas on this subject?

Error is here:
`unable to read property list from file: /Users/testuser/Desktop/bluetooth-ble/ios/bluetoothble/Info.plist: The operation couldn’t be completed. (XCBUtil.PropertyListConversionError error 2.)`

dusker, 2024-05-28:
This is likely unrelated to the library itself but rather to formatting thst plist file. Can you share it here?

avcismail, 2024-05-29:
> This is likely unrelated to the library itself but rather to formatting this plist file. Can you share it here?

The problem was that in the info plist file, the line where I gave the bluetooth permission affected other lines. I solved this problem, thank you.

intent-kacper-cyranowski, 2024-06-06:
Closing as issue seems to be resolved