dotintent/react-native-ble-plx #1206: I can write data to the device, but I cannot receive data.

avcismail, 2024-05-29:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [ ] I'm running the latest version

### Question

Hi, I can write data to the device, but I cannot receive data. Am I using the library methods wrong? Can anyone help?

My code is here:
```
import { StatusBar } from "expo-status-bar";
import React, { useEffect, useState } from "react";
import {
  StyleSheet,
  Text,
  View,
  PermissionsAndroid,
  Button,
  Platform,
} from "react-native";
import { BleManager, State } from "react-native-ble-plx";
import { AppRegistry } from "react-native";
import { name as appName } from "./app.json";

const manager = new BleManager();

const App = () => {
  const [isConnected, setIsConnected] = useState(false);
  const [deviceID, setDeviceID] = useState(null);

  const serviceUUID = "6E400001-B5A3-F393-E0A9-E50E24DCCA9E";
  const characteristicUUIDRX = "6E400002-B5A3-F393-E0A9-E50E24DCCA9E";
  const characteristicUUIDTX = "6E400003-B5A3-F393-E0A9-E50E24DCCA9E";

  const encodeToBase64 = () => {
    try {
      const c1 = String.fromCharCode(14);
      const c2 = "5";
      const str = "reset";
      const sum = c1 + c2 + str;

      const unicodeValues = sum.split("").map((char) => char.charCodeAt(0));
      const dataToWrite = new Uint8Array(unicodeValues);

      return btoa(unescape(encodeURIComponent(sum)));
    } catch (e) {
      console.error("Base64 encoding failed:", e);
      return null;
    }
  };

  useEffect(() => {
    const subscription = manager.onStateChange((state) => {
      if (state === "PoweredOn") {
        scan();
        subscription.remove();
      }
    }, true);
    return () => subscription.remove();
  }, [manager]);

  const writeCharacteristic = async () => {
    if (!deviceID) {
      console.log("Device ID is not set");
      return;
    }
    await manager
      .writeCharacteristicWithResponseForDevice(
        deviceID,
        serviceUUID,
        characteristicUUIDRX,
        encodeToBase64()
      )
      .then((value) => {
        console.log("Write success: ", value);
      })
      .catch((error) => {
        console.log("Write Failed: ", error);
      });
  };

  const monitorCharacteristicData = async () => {
    try {
      const device = await manager.isDeviceConnected(deviceID);
      if (device) {
        console.log("Device is connected");
        await manager.monitorCharacteristicForDevice(
          deviceID,
          serviceUUID,
          characteristicUUIDTX,
          (error, characteristic) => {
            if (error) {
              console.log("Monitor failed:", error);
              return;
            }
            console.log("Received data:", characteristic.value);
          }
        );
      } else {
        console.log("Device is not connected");
      }
    } catch (error) {
      console.log("Error: ", error);
    }
  };

  const scan = () => {
    manager.startDeviceScan(null, null, (error, device) => {
      if (error) {
        console.log(error);
        return;
      }
      console.log(device.name);
      console.log(device.id);
      if (device.name === "YDUART") {
        console.log("Device is found!");
        manager.stopDeviceScan();
        setDeviceID(device.id);
        connect(device.id);
      }
    });
  };

  const connect = async (deviceId) => {
    try {
      await manager
        .connectToDevice(deviceId)
        .then((device) => {
          console.log("Connected to device:", device.name);
          setIsConnected(true);
          console.log("Services and characteristics discovered");
          return device.discoverAllServicesAndCharacteristics();
        })
        .catch((error) => {
          console.error("Error connecting to device:", error);
        });
    } catch (error) {
      console.error("Error connecting to device:", error);
    }
  };

  const disconnectFromDevice = async () => {
    try {
      const device = await manager.isDeviceConnected(deviceID);
      if (device) {
        await manager
          .cancelDeviceConnection(deviceID)
          .then((device) => {
            console.log("Disconnect success:", device);
            setIsConnected(false);
          })
          .catch((error) => {
            console.log("Disconnect Failed: ", error);
          });
      } else {
        console.log("Device is not connected");
      }
    } catch (error) {
      console.log("Error: ", error);
    }
  };

  return (
    <View style={styles.container}>
      <Text>Open up App.js to start working on your app!</Text>
      <StatusBar style="auto" />
      <Button title="Connect to Device" onPress={() => scan()} />
      <Button title="Write to Device" onPress={writeCharacteristic} />
      <Button title="Disconnect from Device" onPress={disconnectFromDevice} />
    </View>
  );
};

AppRegistry.registerComponent(appName, () => App);

const styles = StyleSheet.create({
  container: {
    flex: 1,
    backgroundColor: "#fff",
    alignItems: "center",
    justifyContent: "center",
  },
});

export default App;
```

intent-kacper-cyranowski, 2024-06-06:
Hi, I don't see where are you using yours `monitorCharacteristicData` function.

I manage to make it work after changing your connect function to:
```javascript
  const connect = async deviceId => {
    try {
      await manager
        .connectToDevice(deviceId)
        .then(device => {
          console.log('Connected to device:', device.name)
          console.log('Services and characteristics discovered')
          return device.discoverAllServicesAndCharacteristics()
        })
        .then(() => setIsConnected(true))
        .catch(error => {
          console.error('Error connecting to device:', error)
        })
    } catch (error) {
      console.error('Error connecting to device:', error)
    }
  }
```

and adding `useEffect` to run after successful connection

```javascript
  useEffect(() => {
    if (isConnected) {
      monitorCharacteristicData()
    }
  }, [isConnected])
```

Otherwise you can add additional button to run `monitorCharacteristicData`

intent-kacper-cyranowski, 2024-06-14:
Hi, did you manage to solve your issue?

intent-kacper-cyranowski, 2024-06-20:
Closing due to inactivity

closed 2024-06-20 by intent-kacper-cyranowski