dotintent/react-native-ble-plx #1209: BleError: Device Is Not Connected

annie-hua, 2024-06-05:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

Hi! I have this function 
```
const connectToDevice = async (device: Device) => {
    try {
      const deviceConnection = await bleManager.connectToDevice(device.id);
=      console.log(deviceConnection.id)
      setConnectedDevice(deviceConnection);
      console.log('setConnectedDevice(deviceConnection);')
      await deviceConnection.discoverAllServicesAndCharacteristics();
      // console.log('await deviceConnection.discoverAllServicesAndCharacteristics();')
      bleManager.stopDeviceScan();
      // console.log('Successfully connected')

    } catch (e) {
      console.log("FAILED TO CONNECT", e);
      console.log(e)
    }
  };
```
I see that a `deviceConnection.id` gets printed but when it hits the line `await deviceConnection.discoverAllServicesAndCharacteristics();` I get the following error:
`[BleError: Device 8B16ACE7-C966-13EC-A938-6D2C78E748D9 is not connected]`

I am currently using an iPhone with the app deployed via Expo. I can see that the device is connected via Bluetooth in settings and it shows during the scan. The ID does get printed. What am I missing?

intent-kacper-cyranowski, 2024-06-06:
Hi, looking at the code you have provided I think we could fix it by running `bleManager.stopDeviceScan();` as fast as you find the device you want to connect to. 
In this case you may be running in the situation, where you found the device, you start to connect to it etc. and in the same time you are still scanning for a device and you may match the same device again.

I hope it helps, if issue persists, please provide more context, whole flow ideally.

annie-hua, 2024-06-06:
I made the change for to the code and I'm still facing the same issue: [Change](https://github.com/annie-hua/expo-ble-sample/blob/4ff378149e2b6fb76c442017260e215f99691fc9/useBLE.tsx#L113)

I am running this on Expo with the command `eas build --profile development --platform ios`

intent-kacper-cyranowski, 2024-06-12:
Hi @annie-hua 
I managed to fix the hook by keeping instance of `BleManager` outside of React tree. Declaring `const bleManager = new BleManager()` outside of hook solved the problem. Let me know if it solved your's issue

```typescript
const bleManager = new BleManager()

function useBLE(): BluetoothLowEnergyApi {
```

intent-kacper-cyranowski, 2024-06-20:
Hi @annie-hua 
Do you still need help with the issue?

intent-kacper-cyranowski, 2024-07-01:
Closing due to inactivity

closed 2024-07-01 by intent-kacper-cyranowski

jogoldirus, 2025-01-07:
> Hi @annie-hua I managed to fix the hook by keeping instance of `BleManager` outside of React tree. Declaring `const bleManager = new BleManager()` outside of hook solved the problem. Let me know if it solved your's issue
> 
> ```ts
> const bleManager = new BleManager()
> 
> function useBLE(): BluetoothLowEnergyApi {
> ```

It solve my problem, only this