dotintent/react-native-ble-plx #1212: 🐛 empty known devices list on Android

derekstavis, 2024-06-11:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

`BleManager.devices([id])` should return a list of known devices matching the IDs on the list. 

### Current Behavior

On Android 10, `BleManager.devices([id])` returns a list of known devices matching the IDs on the list. Such "known" devices are devices that have been previously connected to the app.

On Android 10+ (verified only on latest Android though), the list comes back as empty. Calling `BleManager.connectToDevice` successfully connects to the device, and now further calls to `BleManager.devices([id])` succeed. If the app is restarted, the `BleManager.devices([id])` returns empty again.

### Library version

3.2.0

### Device

Galaxy A15

### Environment info

```shell
System:
  OS: macOS 14.5
  CPU: (10) arm64 Apple M1 Pro
  Memory: 96.88 MB / 32.00 GB
  Shell:
    version: 3.7.0
    path: /opt/homebrew/bin/fish
Binaries:
  Node:
    version: 18.19.0
    path: ~/.nvm/versions/node/v18.19.0/bin/node
  Yarn:
    version: 1.22.22
    path: /opt/homebrew/bin/yarn
  npm:
    version: 10.2.3
    path: ~/.nvm/versions/node/v18.19.0/bin/npm
  Watchman:
    version: 2024.04.15.00
    path: /usr/local/bin/watchman
Managers:
  CocoaPods:
    version: 1.15.2
    path: /opt/homebrew/lib/ruby/gems/3.2.0/bin/pod
SDKs:
  iOS SDK:
    Platforms:
      - DriverKit 23.5
      - iOS 17.5
      - macOS 14.5
      - tvOS 17.5
      - visionOS 1.2
      - watchOS 10.5
  Android SDK: Not Found
IDEs:
  Android Studio: 2023.3 AI-233.14808.21.2331.11842104
  Xcode:
    version: 15.4/15F31d
    path: /usr/bin/xcodebuild
Languages:
  Java:
    version: 17.0.10
    path: /opt/homebrew/opt/openjdk@17/bin/javac
  Ruby:
    version: 3.3.0
    path: /opt/homebrew/opt/ruby/bin/ruby
npmPackages:
  "@react-native-community/cli": Not Found
  react:
    installed: 18.2.0
    wanted: 18.2.0
  react-native:
    installed: 0.74.1
    wanted: 0.74.1
  react-native-macos: Not Found
npmGlobalPackages:
  "*react-native*": Not Found
Android:
  hermesEnabled: true
  newArchEnabled: false
iOS:
  hermesEnabled: true
  newArchEnabled: false

info React Native v0.74.2 is now available (your project is running on v0.74.1).
info Changelog: https://github.com/facebook/react-native/releases/tag/v0.74.2
info Diff: https://react-native-community.github.io/upgrade-helper/?from=0.74.1
info For more info, check out "https://reactnative.dev/docs/upgrading?os=macos".
```

### Steps to reproduce

1. Call `BleManager.connectToDevice(id)`
2. Call `BleManager.devices([id])` = returns the matching device
3. Kill the app
2. Call `BleManager.devices([id])` = returns empty list

### Formatted code sample or link to a repository

```js
In steps to reproduce
```

### Relevant log output

```shell
(Will add logs later)
```

madej10, 2024-06-11:
Thanks for bringing this to our attention. We'll look into it as soon as we get the chance 🙏

kacper-cyra, 2024-06-13:
Hi,
 When you call the method `BleManager.devices([id])`, the only thing that it's doing is filtering out devices from the list of scanned devices to check whether there is a device on the list.
 So to make this method works you need to make `BleManager.scanDevices(…)` first so that `BleManager.devices([id])` returns some data.

 Let me know if it is helped you.
 I checked that on Samsung S20 (android 13) I had no issue with returning the list.

derekstavis, 2024-06-14:
After scanning it works as you said. I had to tweak how long the scan runs for in order to get it to work. It seems like 3s is enough. I found it a bit weird that the behavior here is not consistent across iOS and Android. Is that expected?

intent-kacper-cyranowski, 2024-06-14:
I can't say for library authors intentions, but overall it's hard to bring everything under one API and in the same time leverage everything that each of platform can provide when they differ so much.
I will close this issue, in case of more questions fell free to open separate issues

closed 2024-06-14 by intent-kacper-cyranowski