dotintent/react-native-ble-plx #1215: IOS Service UUID does not getting

MAYILERUMPERUMAL, 2024-06-14:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

in ios how to get Service UUID but android is working

intent-kacper-cyranowski, 2024-06-14:
Hi @MAYILERUMPERUMAL 
I'm sorry but I don't really understand what is your question/issue, could you elaborate?

MaxToyberman, 2024-06-19:
I think he means that he `startDeviceScan` doesnt find any devices.

I experience the same issue after upgradie from 2.0.3 to 3.2.0

intent-kacper-cyranowski, 2024-06-19:
Hi @MaxToyberman 
I'm sorry to hear that. Can I point you to our example app? We have there few tests (including device scanning) that works on both iOS and android, maybe you will find there something that will help you. In case on any problems feel free to open an issue with code sample, then I will be able to help you personally.

MaxToyberman, 2024-06-19:
Hi, sure.
Thanks @intent-kacper-cyranowski

MaxToyberman, 2024-06-19:
@intent-kacper-cyranowski @MAYILERUMPERUMAL 

it was my mistake, I was able to fix it by calling stopDeviceScan before starting another scan.

I had 2 processes running in parallel

intent-kacper-cyranowski, 2024-07-01:
Closing as issue author seems inactive

closed 2024-07-01 by intent-kacper-cyranowski

gegana, 2024-10-01:
@intent-kacper-cyranowski this seems to be a bug because the doc states that starting a scan stops an existing scan, yet we actually have to stop it manually.

https://dotintent.github.io/react-native-ble-plx/#blemanagerstartdevicescan