dotintent/react-native-ble-plx #1216: Support sharing JSON data between devices

c-info, 2024-06-20:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [ ] I'm running the latest version

### Question

Is there any plan to support sharing JSON data between devices, enabling communication between phones? I want to connect two phones and share data between devices without using the internet, even if both devices are on different networks.

intent-kacper-cyranowski, 2024-06-20:
Hi @c-info 
To establish basic communication between devices you should use [writeCharacteristicWithResponseForDevice](https://dotintent.github.io/react-native-ble-plx/#blemanagerwritecharacteristicwithresponsefordevice) to write base64 data to the characteristics and [readCharacteristicForDevice](https://dotintent.github.io/react-native-ble-plx/#blemanagerreadcharacteristicfordevice) to read data as base64 from device. You can read more more in introduction section of the documentation. 
After data read you will receive base64 which (in your case) you will have to convert to string, and later on parse as the JSON.

Let me know if this answers your question.

intent-kacper-cyranowski, 2024-07-11:
Hi @c-info
Do you still need help with the issue?

closed 2024-07-17 by intent-kacper-cyranowski