dotintent/react-native-ble-plx #1220: 🐛 BluetoothLE is in unknown state, but the state is PoweredOn

Feffe01, 2024-06-26:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

I expect the iOS device to start scanning devices. Android works just fine.

### Current Behavior

The scan gives me the error.message: BluetoothLE is in unknown state.

### Library version

3.2.0

### Device

iPhone 12 Pro (iOS 17.5.1)

### Environment info

```shell
info Fetching system and libraries information...
(node:39214) Warning: Accessing non-existent property 'padLevels' of module exports inside circular dependency
(Use `node --trace-warnings ...` to show where the warning was created)
System:
  OS: macOS 14.5
  CPU: (8) arm64 Apple M2
  Memory: 145.89 MB / 8.00 GB
  Shell:
    version: "5.9"
    path: /bin/zsh
Binaries:
  Node:
    version: 20.15.0
    path: ~/.nvm/versions/node/v20.15.0/bin/node
  Yarn: Not Found
  npm:
    version: 10.8.1
    path: ~/.nvm/versions/node/v20.15.0/bin/npm
  Watchman:
    version: 2024.06.17.00
    path: /opt/homebrew/bin/watchman
Managers:
  CocoaPods:
    version: 1.15.2
    path: /opt/homebrew/bin/pod
SDKs:
  iOS SDK:
    Platforms:
      - DriverKit 23.5
      - iOS 17.5
      - macOS 14.5
      - tvOS 17.5
      - visionOS 1.2
      - watchOS 10.5
  Android SDK: Not Found
IDEs:
  Android Studio: 2024.1 AI-241.15989.150.2411.11948838
  Xcode:
    version: 15.4/15F31d
    path: /usr/bin/xcodebuild
Languages:
  Java:
    version: 22.0.1
    path: /usr/bin/javac
  Ruby:
    version: 2.6.10
    path: /usr/bin/ruby
npmPackages:
  "@react-native-community/cli": Not Found
  react:
    installed: 18.2.0
    wanted: 18.2.0
  react-native:
    installed: 0.74.2
    wanted: 0.74.2
  react-native-macos: Not Found
npmGlobalPackages:
  "*react-native*": Not Found
Android:
  hermesEnabled: true
  newArchEnabled: false
iOS:
  hermesEnabled: true
  newArchEnabled: false
```

### Steps to reproduce

1. Use the component in a code (I use a token to make a post request and an array of beaconAddresses to find one of the beacons. you can remove the post request part and the token if you want, I don't think that's the problem because the request starts when I find the beacon.)
2. Use "npx expo run:ios --device --configuration Release"
3. Use the button in the app and watch the debug text

### Formatted code sample or link to a repository

```js
import React, { useState, useEffect } from 'react';
import { TouchableOpacity, Text, StyleSheet, ActivityIndicator } from 'react-native';
import { BleManager } from 'react-native-ble-plx';
import { responsiveFontSize } from 'react-native-responsive-dimensions';
import axios from 'axios';

function TimbratureButton({ style, title, token, beaconAddress }) {
    const [isScanning, setIsScanning] = useState(false);
    const [foundDevice, setFoundDevice] = useState(false);
    const manager = new BleManager();
		const [debug, setDebug] = useState("std");

    useEffect(() => {
        return () => {
            if (isScanning) {
                manager.stopDeviceScan();
            }
        };
    }, [isScanning]);

    const sendToken = () => {
        console.log("Sending through axios.post: ", token);
        axios.post("CENSORED", {
            valoriPost: [],
            token: token
        })
    }

    const handleSearch = async () => {
        console.log("isScanning: ", isScanning);
        console.log("token: ", token);
        console.log("beaconAddress: ", beaconAddress);
				setDebug("ScanStart");

        if (!isScanning && token != "" && token != undefined && beaconAddress != [] && beaconAddress != undefined) {
            setIsScanning(true);
            setFoundDevice(false);
            try {
                const state = await manager.state();
                if (state !== 'PoweredOn') {
                    console.error('Bluetooth is not on. Please enable Bluetooth to scan for devices.');
										setDebug("BT-Off")
                    setIsScanning(false);
                    return;
                }

                console.log('Starting scanning... BTSTATE: ' + state);
                manager.startDeviceScan(null, null, (error, device) => {
                    if (error) {
                        console.error('Scan error: ', error.message);
												setDebug(`ScanErr: ${error.message}, state: ${state}`);
                        setIsScanning(false);
                        return;
                    }
                    for(let i = 0; beaconAddress[i]; i++)
                    {
                        if (device.id == beaconAddress[i].MacAddress) {
                            console.log("Beacon found: ", device.id);
														setDebug("BeaconFound");
                            manager.stopDeviceScan();
                            setIsScanning(false);
                            setFoundDevice(true);
                            sendToken();
                        }
                    }
                });

                setTimeout(() => {
										setDebug("Timeout");
                    manager.stopDeviceScan();
                    setIsScanning(false);
                    setFoundDevice(false);
                }, 10000);
            } catch (error) {
								setDebug("BTState?");
                console.error('Error getting Bluetooth state:', error);
                setIsScanning(false);
            }
        }
        else
				{
						setDebug("DataMissing");
            console.error("Device scan didn't start because some data is missing");
				}
    };

    return (
				<>

				<Text>{debug}</Text>
				<Text>|</Text>
				<Text>|</Text>
				<Text>|</Text>
				<Text>|</Text>
				<Text>|</Text>
				<Text>|</Text>
				<Text>|</Text>
				<Text>|</Text>
				
        <TouchableOpacity
            onPress={handleSearch}
            style={[
                styles.button,
                isScanning ? styles.scanningButton : foundDevice ? styles.foundButton : styles.normalButton,
                style
            ]}
            disabled={isScanning}
        >
            {isScanning ? (
                <ActivityIndicator size="small" color="#000000" />
            ) : foundDevice ? (
                <Text style={styles.text}>Beacon trovato!</Text>
            ) : (
                <Text style={styles.text}>{title}</Text>
            )}
        </TouchableOpacity>
				</>
    );
}

const styles = StyleSheet.create({
    button: {
        paddingVertical: 12,
        paddingHorizontal: 20,
        borderRadius: 25,
        alignItems: 'center',
        justifyContent: 'center',
        elevation: 2,  // Android shadow
        shadowOpacity: 0.2,  // iOS shadow
        shadowRadius: 4,
        shadowOffset: { height: 2, width: 0 },
        margin: 10,
    },
    scanningButton: {
        backgroundColor: '#FFFF00', // Yellow when scanning
    },
    foundButton: {
        backgroundColor: '#008000', // Green when a device is found
    },
    normalButton: {
        backgroundColor: '#007AFF', // Blue when normal
    },
    text: {
        color: '#FFFFFF',
        fontSize: responsiveFontSize(2),
        fontWeight: '600',
    },
});

export default TimbratureButton;
```

### Relevant log output

```shell
I don't know how, I'm sorry. I use the debug variable and the console.log to understand what's going on.
```

### Additional information

I'm a beginner. The button works on android and doesn't work on iOS. Searching online and in other reports I understood that the problem is that iOS initially tells you that the bluetooth is in an unknown status and then it updates the status. The problem is that in my code I console.log the error.message and the status together, and the status is always PoweredOn. Furthermore, before starting the scan I check if it is PoweredOn and it would tell me "BTState?" if it wasn't.

intent-kacper-cyranowski, 2024-06-27:
You are correct that on iOS, the initial Bluetooth status is **unknown** and only changes to **PoweredOn** after a short while. We have included this information in our documentation. You can find more details here: [Waiting for PoweredOn State](https://github.com/dotintent/react-native-ble-plx/blob/master/INTRO.md#waiting-for-powered-on-state).

The issue in your code is that you are calling the static method: **manager.state()**.

Instead, you should create a subscription using **manager.onStateChange(state => {}).** When the status changes to PoweredOn, you can start scanning. Please refer to the example function **initializeBLE** in this file: [BLEService.ts](https://github.com/dotintent/react-native-ble-plx/blob/master/example/src/services/BLEService/BLEService.ts).

Hope this helps.

Feffe01, 2024-07-01:
I've tried the following code but it gives me the same problem on iOS. Sorry that it is starting to look like a mess, but meanwhile I'm working on the android version and I'm using a lot of console.logs to understand what's going on.

```
import React, { useState, useEffect } from 'react';
import { TouchableOpacity, Text, StyleSheet, ActivityIndicator, PermissionsAndroid, Platform } from 'react-native';
import { BleManager } from 'react-native-ble-plx';
import { responsiveFontSize } from 'react-native-responsive-dimensions';
import axios from 'axios';

async function requestPermissions() {
    if (Platform.OS === 'android') {
        try {
            const granted = await PermissionsAndroid.requestMultiple([
                PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
                PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION,
                PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN,
                PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT,
            ]);
            const allGranted = Object.values(granted).every(permission => permission === PermissionsAndroid.RESULTS.GRANTED);
            return allGranted;
        } catch (err) {
            console.error(err);
            return false;
        }
    } else {
        return true;
    }
}

function TimbratureButton({ style, title, token, beaconAddress }) {
    const [isScanning, setIsScanning] = useState(false);
    const [foundDevice, setFoundDevice] = useState(false);
    const [bluetoothState, setBluetoothState] = useState('Unknown');
    const manager = new BleManager();
    const [debug, setDebug] = useState("std");

    useEffect(() => {
        const subscription = manager.onStateChange((state) => {
            setBluetoothState(state);
            if (state === 'PoweredOn' && isScanning) {
                startScan();
                subscription.remove();
            }
        }, true);

        return () => {
            subscription.remove();
            if (isScanning) {
                manager.stopDeviceScan();
            }
        };
    }, [isScanning]);

    useEffect(() => {
        const checkPermissions = async () => {
            const hasPermissions = await requestPermissions();
            if (!hasPermissions) {
                console.error('Required permissions not granted.');
                setDebug('Permissions not granted');
            }
        };
        checkPermissions();
    }, []);

    const sendToken = () => {
        console.log("Sending through axios.post: ", token);
        axios.post("CENSORED", {
            valoriPost: [],
            token: token
        });
[... 95 lines cut ...]
        justifyContent: 'center',
        elevation: 2,  // Android shadow
        shadowOpacity: 0.2,  // iOS shadow
        shadowRadius: 4,
        shadowOffset: { height: 2, width: 0 },
        margin: 10,
    },
    scanningButton: {
        backgroundColor: '#FFFF00', // Yellow when scanning
    },
    foundButton: {
        backgroundColor: '#008000', // Green when a device is found
    },
    normalButton: {
        backgroundColor: '#007AFF', // Blue when normal
    },
    text: {
        color: '#FFFFFF',
        fontSize: responsiveFontSize(2),
        fontWeight: '600',
    },
});

export default TimbratureButton;
```

intent-kacper-cyranowski, 2024-07-01:
Ok, so I little simplified your example for faster reproduction and I manage to make it work. Mainly I moved the initialization of `BleManager` outside of React. Let me know if it helped you

```jsx
import React, { useState, useEffect } from 'react'
import { TouchableOpacity, Text, StyleSheet, ActivityIndicator, PermissionsAndroid, Platform } from 'react-native'
import { BleManager } from 'react-native-ble-plx'
// import { responsiveFontSize } from 'react-native-responsive-dimensions'
// import axios from 'axios'

async function requestPermissions() {
  if (Platform.OS === 'android') {
    try {
      const granted = await PermissionsAndroid.requestMultiple([
        PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
        PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION,
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN,
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT
      ])
      const allGranted = Object.values(granted).every(permission => permission === PermissionsAndroid.RESULTS.GRANTED)
      return allGranted
    } catch (err) {
      console.error(err)
      return false
    }
  } else {
    return true
  }
}

const manager = new BleManager() // it should be a singleton (initialized once for whole app), it can't be place inside of React tree (components, hooks etc.)

export const TimbratureButton = ({ style, title, token }) => {
  const [isScanning, setIsScanning] = useState(false)
  const [foundDevice, setFoundDevice] = useState(false)
  const [bluetoothState, setBluetoothState] = useState('Unknown')
  const [debug, setDebug] = useState('std')

  useEffect(() => {
    const subscription = manager.onStateChange(state => {
      setBluetoothState(state)
      // if (state === 'PoweredOn' && isScanning) {
      // start scanning if it's not already running - before caused duplicated scanning
      if (state === 'PoweredOn' && !isScanning) {
        console.log('Bluetooth is powered on, starting scan...')
        startScan()
        subscription.remove()
      }
    }, true)

    return () => {
      subscription.remove()
      if (isScanning) {
        manager.stopDeviceScan()
      }
    }
  }, [isScanning])

  useEffect(() => {
    const checkPermissions = async () => {
      const hasPermissions = await requestPermissions()
      if (!hasPermissions) {
        console.error('Required permissions not granted.')
        setDebug('Permissions not granted')
      }
    }
    checkPermissions()
  }, [])

  const sendToken = () => {
    console.log('Sending through axios.post: ', token)
    // axios.post('CENSORED', {
    //   valoriPost: [],
    //   token: token
    // })
  }

  const startScan = () => {
    console.log('Starting scanning...')
    setDebug('ScanStart')
    manager.startDeviceScan(null, { allowDuplicates: true }, (error, device) => {
      if (error) {
        console.error('Scan error: ', error.message)
        setDebug(`ScanErr: ${error.message}, state: ${bluetoothState}`)
        setIsScanning(false)
        return
      }
      console.log(device.id)
      // for (let i = 0; beaconAddress[i]; i++) {
      // if (device.id === beaconAddress[i].MacAddress) {
      if (device && device.name === 'Galaxy M31') {
        console.log('Beacon found: ', device.name)
        setDebug('BeaconFound')
        manager.stopDeviceScan()
        setIsScanning(false)
        setFoundDevice(true)
        sendToken()
        // break
        // }
      }
    })

    setTimeout(() => {
      setDebug('Timeout')
      manager.stopDeviceScan()
      setIsScanning(false)
      setFoundDevice(false)
    }, 60000)
  }

  const handleSearch = async () => {
    console.log('isScanning: ', isScanning)
    console.log('token: ', token)
    setDebug('ScanInit')

    if (!isScanning && token) {
      setIsScanning(true)
      setFoundDevice(false)

      if (bluetoothState === 'PoweredOn') {
        startScan()
      } else {
        setDebug('Waiting for Bluetooth to power on')
        console.log('Waiting for Bluetooth to power on')
      }
    } else {
      setDebug('DataMissing - isScanning: ' + isScanning + ' token: ' + token)
      console.error("Device scan didn't start because some data is missing")
    }
  }

  return (
    <>
      <Text>{debug}</Text>
      <Text>|</Text>
      <Text>|</Text>
      <Text>|</Text>
      <Text>|</Text>
      <Text>|</Text>
      <Text>|</Text>
      <Text>|</Text>
      <Text>|</Text>

      <TouchableOpacity
        onPress={handleSearch}
        style={[
          styles.button,
          isScanning ? styles.scanningButton : foundDevice ? styles.foundButton : styles.normalButton,
          style
        ]}
        disabled={isScanning}
      >
        {isScanning ? (
          <ActivityIndicator size="small" color="#000000" />
        ) : foundDevice ? (
          <Text style={styles.text}>Beacon trovato!</Text>
        ) : (
          <Text style={styles.text}>{title}</Text>
        )}
      </TouchableOpacity>
    </>
  )
}

const styles = StyleSheet.create({
  button: {
    paddingVertical: 12,
    paddingHorizontal: 20,
    borderRadius: 25,
    alignItems: 'center',
    justifyContent: 'center',
    elevation: 2, // Android shadow
    shadowOpacity: 0.2, // iOS shadow
    shadowRadius: 4,
    shadowOffset: { height: 2, width: 0 },
    margin: 10
  },
  scanningButton: {
    backgroundColor: '#FFFF00' // Yellow when scanning
  },
  foundButton: {
    backgroundColor: '#008000' // Green when a device is found
  },
  normalButton: {
    backgroundColor: '#007AFF' // Blue when normal
  },
  text: {
    color: '#FFFFFF',
    // fontSize: responsiveFontSize(2),
    fontSize: 20,
    fontWeight: '600'
  }
})

```

Feffe01, 2024-07-02:
Perfect, now it's working fine. Thank you very much.
Can I ask you one more thing? On Android it was working even before this fix and the console.log(device.id) was printing all the devices he was finding. The problem I have now is that it doesn't find MY ble beacon, even if I check every single Mac address and even if I let him scan for minutes. I can't really understand why my android 11 finds it in few seconds with the same app, while my android 14 and now my iOS 17.5.1 doesn't. The main differences I found are the permissions but I don't think this is the case because the scan shouldn't start at all. Do you have any idea of some reasons that can cause this? I have thought about incompatible bluetooth, but, correct me if I'm wrong, newer bluetooth versions can scan older ones.

This is the ble beacon I'm testing on:
[amazon.it link](https://www.amazon.it/Ibeacons-Bluetooth-Nrf51822-Custodia-Silicone/dp/B07HG62PCJ/ref=sr_1_18_sspa?__mk_it_IT=ÅMÅŽÕÑ&crid=2KC0E4TFIBXTU&dib=eyJ2IjoiMSJ9.msHA7FgH2jvcM8N3JtpIdf8GNBywexhLPu4uFFwLzyxXQdg3MkDNmrJu4yJ8_uhthsT0moA2Uj5_Vnj_IxWcvJ7QsdUIMUff6xHhlAxRfaCi5W3uf6s2QuRhpexsR0raRNfRfCckqoueCQAaQjXImVakQweWqjzJLUWHifTLxzCimMQccikX_D5D1uxiwqiGp3uHx94kBSqBSrCPwHuq3YPYhcoB3o1s2ILscpt58Qs_By6OoPCesYym7Wl_kICWKMP4I84KebVtYU4dUmG12pZE9_f85oGtTqZ_oNaoYWs.T82rWFVP2G2MboRGnz1Wp9WSGoYY4oGMeqjzcX8Y9Jo&dib_tag=se&keywords=ble+beacon&qid=1719907650&sprefix=ble+beac%2Caps%2C82&sr=8-18-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9tdGY&psc=1)

I know it's not a good beacon, and it has not docs, but I'm not the one who chose it and, anyway, it can be found by my android 11.

Thank you so much again.
Meanwhile I'm still searching solutions, if I find it I'll close the issue.

intent-kacper-cyranowski, 2024-07-02:
You are right that permissions won’t matter.

There might be many factors that influence that. The basic one:
- device is already connected to OS - make sure that you are not connected with device through OS, because if you are then device won’t be discoverable through the app and you would have to use `connectedDevices` to find them
- not closing connection properly - if you kill the app without closing connection, and reopen app, the device you are trying to connect to will still have it’s connection open (even though you lost it) and because of that won’t be discoverable

I’m not sure about it but you may try setting [legacyScan](https://dotintent.github.io/react-native-ble-plx/#scanoptionslegacyscan), though it's android only. You may also want to read about advertising in ble, maybe you will find something you can relate to.

Let me know if any of these helps

Feffe01, 2024-07-03:
I used `connectedDevices` and `isDeviceConnected` to check if the beacon was already connected and it is not. I don't think the problem is the second one you mentioned because with BLEScanner I can see that the beacon is not already connected to anything (If it was, I shouldn't be able to connect to it). Btw to be sure about this, I connected and disconnected it through BLEScanner before activating the scan, but nothing changed.
`legacyScan` is true by default. I've tried `{legacyScan: false, scanMode: 'LowLatency'}` and different scanMode(s) but no result.
I'm sorry to bother you, I'm trying to fix this for many days and now I don't even know where to turn. I really can't understand why on android 11 it finds my beacon while on iOS and android 14 it is not. Same app, same code, all permissions. Can it be a problem the fact that I'm using device.id instead of device.name or UUIDs? I don't think so because for each device I use `console.log("Detected device:", device.id, device.name, device.rssi);` and it finds them even on android 14.

Thank you again.

Feffe01, 2024-07-04:
I'll create another issue and close this one, because I think that now the problem in unrelated to the main issue.
Thank you for helping me solve this issue.

closed 2024-07-04 by Feffe01