dotintent/react-native-ble-plx #1222: Auto-disconnection in some Android devices

joanmarcs, 2024-07-04:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

Maintain the connection

### Current Behavior

Devices disconnect after some seconds

### Library version

3.2.0

### Device

Samsung s22 ultra, Pixel, ..

### Environment info

```shell
-
```

### Steps to reproduce

-

### Formatted code sample or link to a repository

```js
-
```

### Relevant log output

```shell
-
```

### Additional information

-

joanmarcs, 2024-07-04:
Hi @intent-kacper-cyranowski !

As you said to me, I have opened a new issue. I am attaching the logs you asked me in https://github.com/dotintent/react-native-ble-plx/issues/1197#issuecomment-2204467303

**EXPECTED BEHAVIOUR: the app connects to the device and ask for his state until the user closes the BLE connection manually**

[log.txt](https://github.com/user-attachments/files/16097138/log.txt)

**BAD BEHAVIOUR: the app connects to the device and ask for his state but connection drops**

[log2.txt](https://github.com/user-attachments/files/16097239/log2.txt)

intent-kacper-cyranowski, 2024-07-10:
Hi @joanmarcs 

We have opened a ticket for your issue. I would like to clarify some information. 
Did the error occurred on different devices then previously specified? Samsung and Pixel have their own SoC, and if error occurs only there it might be the cause.
You have mentioned that you are working with vehicle chargers, can you specify what is the device, or can you provide any other device on which you were able to reproduce the issue.

Everything that will help in reproducing and solving the issue is welcome.

ovesco, 2024-07-12:
Also observed with an ESP32 device.

nhaduongc, 2024-07-15:
Also get event disconnect after connect around 10-15 seconds, It's crazy

jakecurreri, 2024-11-22:
+1 @nhaduongc same here. Every few seconds, Android drops connection, reconnects, then drops connection again.

jakecurreri, 2024-11-22:
Solved with this workaround: https://github.com/dotintent/react-native-ble-plx/issues/1197#issuecomment-2256601221

closed 2025-01-08 by aliberski