dotintent/react-native-ble-plx #1227: Read characteristic in background?

lucasmrau, 2024-07-20:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

Someone already tested using with background-fetch to read from a device connected in foreground?

contactsimonwilson, 2024-08-15:
I'm not using background fetch but I am using react-native-background actions to keep a task running indefinitely on iOS and Android that pulls data from a ble device at roughly 15Hz and also subscribes to location updates.

lucasmrau, 2024-08-27:
Are you sending this to a database or saving locally in a state?

lucasmrau, 2024-08-27:
> I'm not using background fetch but I am using react-native-background actions to keep a task running indefinitely on iOS and Android that pulls data from a ble device at roughly 15Hz and also subscribes to location updates.

Are you sending this to a database or saving locally in a state?

@contactsimonwilson

aliberski, 2025-01-13:
It should work with react-native-background-actions. Similar case https://github.com/dotintent/react-native-ble-plx/issues/1172