dotintent/react-native-ble-plx #1228: isDeviceConnected return BleErrorCode.DeviceDisconnected instead of false🐛

philbourg, 2024-07-30:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

if (bleManager.isDeviceConnected(device.id)) should return false when is not connected.

### Current Behavior

if (bleManager.isDeviceConnected(device.id)) throw the error BleErrorCode.DeviceDisconnected when not connected.

### Library version

3.2.1

### Device

iPhone SE (iOS 17.5.1)

### Environment info

```shell
System:
  OS: macOS 13.6.6
  CPU: (12) arm64 Apple M2 Pro
  Memory: 76.55 MB / 16.00 GB
  Shell:
    version: "5.9"
    path: /bin/zsh
Binaries:
  Node:
    version: 18.20.3
    path: ~/.nvm/versions/node/v18.20.3/bin/node
  Yarn:
    version: 1.22.21
    path: /usr/local/bin/yarn
  npm:
    version: 10.7.0
    path: ~/.nvm/versions/node/v18.20.3/bin/npm
  Watchman:
    version: 2024.06.17.00
    path: /usr/local/bin/watchman
Managers:
  CocoaPods:
    version: 1.15.2
    path: /usr/local/bin/pod
SDKs:
  iOS SDK:
    Platforms:
      - DriverKit 23.2
      - iOS 17.2
      - macOS 14.2
      - tvOS 17.2
      - visionOS 1.0
      - watchOS 10.2
  Android SDK: Not Found
IDEs:
  Android Studio: 2023.3 AI-233.14808.21.2331.11842104
  Xcode:
    version: 15.2/15C500b
    path: /usr/bin/xcodebuild
Languages:
  Java:
    version: 17.0.10
    path: /usr/bin/javac
  Ruby:
    version: 2.6.10
    path: /usr/bin/ruby
npmPackages:
  "@react-native-community/cli": Not Found
  react:
    installed: 18.2.0
    wanted: 18.2.0
  react-native:
    installed: 0.73.6
    wanted: 0.73.6
  react-native-macos: Not Found
npmGlobalPackages:
  "*react-native*": Not Found
Android:
  hermesEnabled: true
  newArchEnabled: false
iOS:
  hermesEnabled: true
  newArchEnabled: false
```

### Steps to reproduce

1. Scan for devices 
2. Memorize the device 
3. Call your connectToDevice function 
4. Add a isDeviceConnected(device.id) before connecting your device 
5. Verify if you have an error instead of false 

### Formatted code sample or link to a repository

```js
const connectToDevice = async (device) => {

try {
      if (bleManager.isDeviceConnected(device.id)) {
        await bleManager.cancelDeviceConnection(device.id);
      }
    } catch (error) {

      if(error.errorCode === BleErrorCode.DeviceDisconnected) {
      vConsoleLog(
        fileName,
        `Error while disconnecting device: ${error.message}`
      );
    }

   // Connect your device if no error 
}
```

### Relevant log output

```shell
🟢 Creating JS object for module 'NativeModulesProxy'
🟢 Creating JS object for module 'ExponentConstants'
🟢 Creating JS object for module 'ExponentFileSystem'
🟢 Creating JS object for module 'ExpoDevice'
Unbalanced calls start/end for tag 19
Running "main" with {"rootTag":11,"initialProps":{}}
🟢 Creating JS object for module 'ExpoKeepAwake'
API MISUSE: <CBCentralManager: 0x300c68a00> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
--> BLE permissions granted
--> State Manager : Default
[GESTURE HANDLER] Initialize gesture handler for view <RCTRootContentView: 0x10690ca60; frame = (0 0; 375 667); gestureRecognizers = <NSArray: 0x3033ad980>; layer = <CALayer: 0x303dc6ea0>> reactTag: 11; frame = {{0, 0}, {375, 667}}; layer = <CALayer: 0x303dc6ea0>
--> Start scan devices
PoweredOn
--> Device scanned:  MYDEVICENAME MYDEVICEID
--> new device scanned:  MYDEVICENAME
quic_conn_process_inbound [C57.1.1.1:2] [-cccc99356eb95e52] unable to parse packet
--> Initialize app 0,2,29
--> Device scanned:  MYDEVICENAME MYDEVICEID
--> new device scanned:  MYDEVICENAME
--> Stop scan devices
--> device id MYDEVICEID
--> State Manager : Connecting
--> Error while disconnecting device BleError: Operation was cancelled
```

### Additional information

Bonjour! 

I use the PLX library since January. Great library, really like it, thanks for the good work.

I had to add this verification because after calling : 

```
deviceConnection = await bleManager.connectToDevice(device.id, {
          autoConnect: false,
        });
```

I know that is a weird verification to do before connecting, but I get a deviceConnection.name and deviceConnection.localName to null after calling bleManager.connectToDevice (I will write another Issues for this one) and I try to understand if my bleManager is disconnecting correctly my device when I call bleManager.onDeviceDisconnected. 

Thanks for you help.

intent-kacper-cyranowski, 2024-07-31:
Hi @philbourg 
Thank you for kind words 😄 

I can't reproduce the behaviour with the provided code.

```ts
  useEffect(() => {
    bleManager.startDeviceScan(null, null, async (error, device) => {
      if (device && device.name === 'someDeviceName') {
        bleManager.stopDeviceScan()
        connectToDevice(device)
      }
    })
  }, [])

  const connectToDevice = async (device: Device) => {
    try {
      if (bleManager.isDeviceConnected(device.id)) {
        await bleManager.cancelDeviceConnection(device.id)
      }
    } catch (error) {
      if (error.errorCode === BleErrorCode.DeviceDisconnected) {
        console.log(`Error while disconnecting device: ${error.message}`)
      }
    }

    device
      .connect()
      .then(device => {})
      .catch(error => {
        console.error('error', error)
      })
  }
```

I have noticed one issue in provided code. The `isDeviceConnected()` method is async so you need to await and this may be the couse of the error as Promise is truthful so the `cancelDeviceConnection` is always being run. The final code should look like:

```ts
      if (await bleManager.isDeviceConnected(device.id)) {
        await bleManager.cancelDeviceConnection(device.id)
      }
```

I have also question on what is the case for handling already connected devices by disconnecting them. Why you just don't ignore already connected devices? Do you have multiple devices you want to connect to at once? Handling this case properly may remove the need for checking connection state.

I hope it helps. If issue persist please provide more context of how you use the library:
- how you initialize `BleManager`
- how you scan for devices
- how you connect to device

philbourg, 2024-07-31:
Hi @intent-kacper-cyranowski ! :) 

Adding the _await_ resolve it!

Thanks a lot! 

> I have also question on what is the case for handling already connected devices by disconnecting them. Why you just don't ignore already connected devices? Do you have multiple devices you want to connect to at once? Handling this case properly may remove the need for checking connection state.

I want to be sure that I have only one and one device at the time connected for now.  It's more a QA function for my app, I want to detect if I had problem with cancelDeviceConnection.

closed 2024-07-31 by philbourg