dotintent/react-native-ble-plx #1230: Issue with [BleError: Operation was cancelled] on iOS 17

HwangJungeon, 2024-08-04:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

Hello,

I am encountering an issue with the **[BleError: Operation was cancelled]** error on iOS 17. This problem does not occur when testing on AOS, but it started appearing on iOS after a period of smooth operation. The issue persists consistently.

I have reviewed and attempted solutions suggested in previous issues, such as [Connection Error 2 "Operation was cancelled" on IOS16 #1080][1]. I tried using `device.cancelConnection()` before `device.connect()` and modified my code to ensure that `new BleManager()` is called only once by implementing a singleton pattern. Despite these efforts, the problem remains unresolved.

The most perplexing aspect is that the connection works fine on iOS initially, but the error reappears the next day upon retrying.

Additionally, I have verified iOS permissions, cleaned the iOS build folder, and retried the process. I am currently using the latest version of `react-native-ble-plx` (3.2.1).

**Environment:**
- iOS Version: 17
- react-native-ble-plx Version: 3.2.1

Any assistance or insights into resolving this issue would be greatly appreciated.

Thank you.

[1]: https://github.com/dotintent/react-native-ble-plx/issues/1080

shanez, 2024-08-21:
@HwangJungeon What iOS minor version are you on?

HwangJungeon, 2024-08-22:
> @HwangJungeon What iOS minor version are you on?

I used iOS version 17.5.1 (21F90)

ebadia, 2024-08-23:
Same here using 3.2.1 on:
- iOS 17.5.1
- iOS 16.7.8

Tried several "connect/disconnect" combinations all withe the same result:
`'ERROR', [BleError: Operation was cancelled]`

Help appreciated too. Thanks!!!

HwangJungeon, 2024-08-24:
@ebadia 
In my case, I tried reinstalling the node module, which helped a bit, but still intermittently throwing the error 'ERROR', `[BleError: Operation was canceled]`

ebadia, 2024-08-25:
> @ebadia In my case, I tried reinstalling the node module, which helped a bit, but still intermittently throwing the error 'ERROR', `[BleError: Operation was canceled]`

I deleted node_modules folder, install them again, but the error persists. Thanks for your response!

HwangJungeon, 2024-08-25:
> > @ebadia In my case, I tried reinstalling the node module, which helped a bit, but still intermittently throwing the error 'ERROR', `[BleError: Operation was canceled]`
> 
> I deleted node_modules folder, install them again, but the error persists. Thanks for your response!

It still happens to me too... I'll continue to try different things to fix the error, and if I find a good solution, I'll share it!

intent-kacper-cyranowski, 2024-08-26:
Hi @HwangJungeon @ebadia 
Could you provide repro of your issue and collect [logs](https://dotintent.github.io/react-native-ble-plx/#collecting-native-logs)?

ebadia, 2024-08-30:
This is the code I use to identify and connect to the BLE device (a weight scale)

```
            if (device.name === "Chipsea-BLE") {
              console.log("IDENTIFICADOR", device.name);
              console.log("IM CONNECTED", await device.isConnected());

              ble.stopDeviceScan();
              if (!await device.isConnected()) {
                try {
                  await device.connect();

                  console.log("CONNECTED", device.id);

                } catch (e) {
                  console.log("ERROR CONN", e);
                }
              }
```

This is the log from this piece of code:
```
[RxBLEKit|VERB|12:05:52.175]: CentralManager(10763401600) didDiscover(peripheral: Peripheral(uuid: 1ECEA624-9BC7-60C4-74B4-E09FA61E3456, name: Optional("Chipsea-BLE")),
rssi: -60)
2024-08-30 13:05:52.176105+0100 pronokalPre[38603:3770084] [javascript] 'IDENTIFICADOR', 'Chipsea-BLE'
2024-08-30 13:05:52.178016+0100 pronokalPre[38603:3770084] [javascript] 'IDENTIFICADOR', 'Chipsea-BLE'
[RxBLEKit|VERB|12:05:52.178]: CentralManager(10763401600) didDiscover(peripheral: Peripheral(uuid: 3E3349C0-05D8-8F5C-E6CF-9DA14F1D790C, name: nil),
rssi: -73)
2024-08-30 13:05:52.179856+0100 pronokalPre[38603:3770084] [javascript] 'IM CONNECTED', false
[RxBLEKit|VERB|12:05:52.180]: CentralManager(10763401600) didDiscover(peripheral: Peripheral(uuid: 3E3349C0-05D8-8F5C-E6CF-9DA14F1D790C, name: nil),
rssi: -73)
[RxBLEKit|DEBG|12:05:52.182]: CentralManager(10763401600) stopScan()
2024-08-30 13:05:52.183974+0100 pronokalPre[38603:3770084] [javascript] 'IM CONNECTED', false
[RxBLEKit|DEBG|12:05:52.186]: CentralManager(10763401600) retrievePeripherals(
withIdentifiers: [1ECEA624-9BC7-60C4-74B4-E09FA61E3456])
[RxBLEKit|DEBG|12:05:52.190]: CentralManager(10763401600) connect(
peripheral: Peripheral(uuid: 1ECEA624-9BC7-60C4-74B4-E09FA61E3456, name: Optional("Chipsea-BLE")),
options: nil)
[RxBLEKit|DEBG|12:05:52.191]: CentralManager(10763401600) retrievePeripherals(
withIdentifiers: [1ECEA624-9BC7-60C4-74B4-E09FA61E3456])
[RxBLEKit|DEBG|12:05:52.191]: CentralManager(10763401600) connect(
peripheral: Peripheral(uuid: 1ECEA624-9BC7-60C4-74B4-E09FA61E3456, name: Optional("Chipsea-BLE")),
options: nil)
[RxBLEKit|DEBG|12:05:52.192]: CentralManager(10763401600) cancelPeripheralConnection(
peripheral: Peripheral(uuid: 1ECEA624-9BC7-60C4-74B4-E09FA61E3456, name: Optional("Chipsea-BLE")))
[RxBLEKit|DEBG|12:05:52.226]: CentralManager(10763401600) didDisconnect(from: Peripheral(uuid: 1ECEA624-9BC7-60C4-74B4-E09FA61E3456, name: Optional("Chipsea-BLE")),
error: nil)
2024-08-30 13:05:52.229912+0100 pronokalPre[38603:3770084] [javascript] 'ERROR CONN', [BleError: Operation was cancelled]
[RxBLEKit|DEBG|12:06:05.973]: CentralManager(10763401600) cancelPeripheralConnection(
peripheral: Peripheral(uuid: 1ECEA624-9BC7-60C4-74B4-E09FA61E3456, name: Optional("Chipsea-BLE")))
2024-08-30 13:06:06.010458+0100 pronokalPre[38603:3770081] [CoreBluetooth] XPC connection invalid
```

Hope it helps you!

HwangJungeon, 2024-09-02:
@intent-kacper-cyranowski 

I apologize for the delayed response to your comment. As the issue persists, I've attempted the following approach, which seems to work without any problems:

```javascript
if (error instanceof Error && error.message.includes('Operation was cancelled')) {
    this.manager.destroy();
    this.manager = new BleManager();
    return this.connectToDevice(deviceId);
}
```

This method catches the "Operation was cancelled" error, destroys the existing instance, creates a new one, and then attempts to reconnect to the device.

Do you think this approach is acceptable to use?

ebadia, 2024-09-09:
> @intent-kacper-cyranowski
> 
> I apologize for the delayed response to your comment. As the issue persists, I've attempted the following approach, which seems to work without any problems:
> 
> ```js
> if (error instanceof Error && error.message.includes('Operation was cancelled')) {
>     this.manager.destroy();
>     this.manager = new BleManager();
>     return this.connectToDevice(deviceId);
> }
> ```
> 
> This method catches the "Operation was cancelled" error, destroys the existing instance, creates a new one, and then attempts to reconnect to the device.
> 
> Do you think this approach is acceptable to use?

Still have issues, will continue trying different approaches. Thanks!

zachTrio, 2024-10-15:
> Still have issues, will continue trying different approaches. Thanks!

Have you settled on a new approach? Running into the same error.

HwangJungeon, 2024-10-15:
@zachTrio The method I showed is simple, but still works well. I'm still using it and it's working fine on iOS 18.

Akuma-U1, 2024-10-29:
Hi, I encountered this error as well, and fixed it. The error comes from iOS scanning very quickly in the `startDeviceScan` function. The listener callback function activates multiple times for the same device id, causing the device to cancel all of the connections.

To fix this, you can do the following:
1. Explicitly tell the BLEManger to not catch duplicates in the `options` of the `startDeviceScan` function, as follows
```typescript
bleManager.startDeviceScan(
        [SERVICE_UUID_LIST],
        {
          allowDuplicates: false, // <----- This line
          callbackType: ScanCallbackType.AllMatches,
          scanMode: ScanMode.LowLatency
        },
        deviceFoundFunction
)
```

2. Inside of the listener function you can check wether or not the current device you are trying to connect to, is already trying to get a connection. I did this by calling `startDeviceScan()` in a wrapper function, then having a variable called `currentDeviceId` at the top level of the function, then setting the listener function as an arrow function so that it can have `currentDeviceId` in its scope. The implementation looks something like this

```typescript
function startDeviceScanWrapper(deviceName:string){
   let currentDeviceId = ""; // Set variable that keeps track of what we are busy with
   bleManager.startDeviceScan( [SERVICE_UUID_LIST],
        {
          allowDuplicates: false, 
          callbackType: ScanCallbackType.AllMatches,
          scanMode: ScanMode.LowLatency
        },
        async (error, device) => { // <------ Start of device scan callback listener
          if (error) {
            await logger(`🛑: Error scanning for devices: ${error}`);
            return;
          }

          // Check if we got a device, and if it's the correct one
          if (
            device &&
            device.isConnectable &&
            (device.name === deviceName || device.localName === deviceName)
          ) {
            const name = device.name ?? device.localName;
            await logger(`✅: Found device: ${name}`);
            if (currentDeviceId === device.id) {
              await logger(
                "🛑: Multiple connection attempts to a single device. Dropping this attempt."
              );
              return;
            } else {
              currentDeviceId = device.id;
            }
            // ..... Rest of your code after this to handle connecting to a device
            // Something like device.connect(......)
       }
}
```

Hope this helps!

george-eucare, 2024-11-21:
> Hi, I encountered this error as well, and fixed it. The error comes from iOS scanning very quickly in the `startDeviceScan` function. The listener callback function activates multiple times for the same device id, causing the device to cancel all of the connections.
> 
> To fix this, you can do the following:
> 
> 1. Explicitly tell the BLEManger to not catch duplicates in the `options` of the `startDeviceScan` function, as follows
> 
> ```ts
> bleManager.startDeviceScan(
>         [SERVICE_UUID_LIST],
>         {
>           allowDuplicates: false, // <----- This line
>           callbackType: ScanCallbackType.AllMatches,
>           scanMode: ScanMode.LowLatency
>         },
>         deviceFoundFunction
> )
> ```

> 
> Hope this helps!

allowDuplicates: false, ---> it works for me,  thank you !

jakecurreri, 2024-11-25:
Issue only with production iOS builds. Cannot replicate locally.

`allowDuplicates` is set to `false`, and I've implemented a fail-safe based on some of the solutions above:

Inside my `connectToDevice`:

```
if (
   error instanceof Error &&
   error.message.includes('Operation was cancelled')
) {
   this.manager.destroy().then(() => {
   this.manager = new BleManager();
      this.connectToDevice(deviceId);
   });
}
```

With `poweredOn` state handling, this works, but I do see a `Unknown error occurred. This is probably a bug!...` in the logs.

closed 2025-01-18 by aliberski