dotintent/react-native-ble-plx #1242: Feature Request: bridge Le Bluetooth Device Address from native code to RN

folin03, 2024-10-16:
### Feature Description

provide Le Bluetooth Device Address

 the address is provided in general advertising data for BLE but the package does not expose in in scanned devices

### Current Workaround

MAC address must be send via manufacturers data

### Suggested Implementation

bridge the LeBluetoothDeviceAddress from ble advertising data from native code to React Navive side.

### Additional Information

- [X] I have searched for existing feature requests.
- [ ] I would like to work on this feature myself.

aliberski, 2025-01-16:
Hello @folin03 . Thank you for raising this topic. I'll add it to the backlog for future enhancements. If you'd like to contribute, pull requests are always welcome!

aliberski, 2025-01-18:
Hello @folin03! On Android, `device.id` already provides the Bluetooth Device Address (MAC address), so you can use that directly. On iOS, however, Apple does not expose the MAC address due to privacy restrictions. In this case, `device.id` provides a unique identifier for the device, which serves a similar purpose. Let us know if there's a specific need that device.id doesn't address, and we can explore further!

closed 2025-01-18 by aliberski

folin03, 2025-01-19:
LE BLUETOOTH DEVICE ADDRESS is Bluetooth standard parameter. I believe that the package should allow access to all standard BLE parameters.

1.16 LE BLUETOOTH DEVICE ADDRESS
1.16.1 Description
The LE Bluetooth Device Address data type defines the device address of the
local device and the address type on the LE transport.

Data Type Description
«LE Bluetooth Device Address» uint48 followed by boolean

The first value is the Device Address
defined in [Vol 6] Part B, Section 1.3.
The second value indicates whether the
Device Address is a Public Address or a
Random Address:
0 = Public Device Address
1 = Random Device Address.

The use case is if there is a number of devices that connect to the app via BLE and also to each other, The logic common id in both systems is MAC address of the device. UUID on iOS is useful but in scenario above not efficient enough since the devices do not know their UUID on iOS device.