dotintent/react-native-ble-plx #1246: Handling Parallel Notifications from Multiple Bluetooth Devices in Real-Time

viswa-capgemini, 2024-10-30:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

Hi,
I am connected to two devices and subscribed to their notifications simultaneously. However, I'm currently only receiving notifications from one device at a time. I need to receive data from both devices in parallel. Is this possible? If so, please share the solution here.

perifit, 2024-10-30:
@viswa-capgemini I'm using this library and success to connect 2 devices at the same time and subscribe multiple chs.
I also experienced the same issue with yours, the solution that I applied in my project: make sure the transaction id is unique for a device and a characteristics.

viswa-capgemini, 2024-10-30:
> hello @viswa-capgemini have subscribed to multiple services from multiple devices?

I am connecting to multiple devices simultaneously that share the same service UUID and characteristic ID and retrieving data from each device through notifications in parallel. The issue I'm facing is that I'm only able to receive data from one device at a time. I will try with @perifit solution and update the result. Thanks

viswa-capgemini, 2024-10-30:
> @viswa-capgemini I'm using this library and success to connect 2 devices at the same time and subscribe multiple chs. I also experienced the same issue with yours, the solution that I applied in my project: make sure the transaction id is unique for a device and a characteristics.

Hi @perifit, Thank you for the suggestion! I'll give it a try and will let you know the results.

viswa-capgemini, 2024-10-30:
> @viswa-capgemini are you having multiple service UUID? for example`const monitorMultipleCharacteristics = async () => { try {
> 
> ```
>   const monitorPromises = uuids.map(({ serviceUUID, characteristicUUID }) => {
> 
>     return new Promise((resolve, reject) => {
>       BLEService.monitorCharacteristicForDevice(
>         serviceUUID,
>         characteristicUUID,
>         async data => {
> 
>             case UUID:
>               const temperature = parse(data.value)
>               break
> 
>             case UUID:
>               console.log('dispatch service')
>               break
> 
>             default:
>               console.log('Unknown UUID', data.uuid)
>           }
>           setTimeout(resolve, 100, data)
>           console.info('success', new Date().getTime())
>         },
>         async error => {
>           console.error(error)
>           await BLEService.finishMonitor()
>           reject()
>         }
>       )
>       // Store the subscription so it can be cleaned up later
>       //subscriptions.push(subscription)
>     })
> 
>     // Wait for each monitor to be set up
>   })
> 
>   Promise.all(monitorPromises)
>     .then(data => {
>       console.log(data)
>     })
>     .catch(error => {
>       console.error('Error in one of the monitors:', error)
>     })
> } catch (error) {
>   console.error('Error setting up characteristic monitoring:', error)
> }
> 
> // Optionally return the subscriptions if you need to stop monitoring later
> //return subscriptions
> ```
> 
> }`

@smitha-2020, In my case we have specified with single serviceUUID, characteristicsUUID. In that case 2 devices have same UUID's. Adding to this, we are subscribing notifications for one particular serviceUUID.

viswa-capgemini, 2024-10-30:
> @viswa-capgemini How are you subscribing to notifications? is it using monitorCharacteristicForDevice?

@smitha-2020. Yes I am using it
const enableNotifications = async () => {
        sub = manager.monitorCharacteristicForDevice(
            deviceID,
            serviceUUID,
            notificationUUID,
            handleNotificationData
        );
}