dotintent/react-native-ble-plx #1248: 🐛 IOS 17.7 and IOS 18.0.1 not able to read ble device characteristics.

sumitsharma7667, 2024-11-12:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

It should show device characterstics

### Current Behavior

It not showing device characterstics

### Library version

3.2.1

### Device

iphone 14 

### Environment info

```shell
System:
    OS: macOS 14.6.1
    CPU: (10) arm64 Apple M1 Pro
    Memory: 89.20 MB / 16.00 GB
    Shell: 5.9 - /bin/zsh
  Binaries:
    Node: 20.11.0 - ~/.nvm/versions/node/v20.11.0/bin/node
    Yarn: 1.22.22 - ~/.nvm/versions/node/v20.11.0/bin/yarn
    npm: 10.2.4 - ~/.nvm/versions/node/v20.11.0/bin/npm
    Watchman: 2024.10.21.00 - /opt/homebrew/bin/watchman
  Managers:
    CocoaPods: 1.15.2 - /opt/homebrew/bin/pod
  SDKs:
    iOS SDK:
      Platforms: DriverKit 23.5, iOS 17.5, macOS 14.5, tvOS 17.5, visionOS 1.2, watchOS 10.5
 IDEs:
    Android Studio: 2024.2 AI-242.23339.11.2421.12483815
    Xcode: 15.4/15F31d - /usr/bin/xcodebuild
  Languages:
    Java: 17.0.13 - /usr/bin/javac
  npmPackages:
    @react-native-community/cli: Not Found
    react: 18.2.0 => 18.2.0 
    react-native: 0.71.11 => 0.71.11 
    react-native-macos: Not Found
  npmGlobalPackages:
    *react-native*: Not Found
```

### Steps to reproduce

1 - start device scan.
2 - read characteristics.
3- decode  characteristics

### Formatted code sample or link to a repository

```js
async function scanBlePlxDevicesWithRetry(UUID: string, desk: any, maxRetries = 3, retryDelay = 1000) {
        let attempt = 0;
        let success = false;

        while (attempt < maxRetries && !success) {
            try {
                attempt++;
                await scanBlePlxDevices(UUID, desk);
                success = true;
            } catch (error) {
                console.warn(`Attempt ${attempt} failed: ${error.message}`);
                if (attempt < maxRetries) {
                    console.log(`Retrying in ${retryDelay}ms...`);
                    await new Promise(resolve => setTimeout(resolve, retryDelay));
                } else {
                    Alert.alert(JSON.stringify(error));
         
                    await bleManager.destroy();
                }
            }
        }
    }

    async function scanBlePlxDevices(UUID: string, desk: any) {
        setUuid(UUID);

        return new Promise(async (resolve, reject) => {
            let device: Device | null = null;

            const scanSubscription = await bleManager.startDeviceScan(null, null, async (error, bleDevice) => {
                if (error) {
                    await bleManager.stopDeviceScan();
                    setCapError(JSON.stringify(error));
                    return reject(new Error("Device scan failed"));
                }

                if (bleDevice?.name?.startsWith("hbd1_")) {
                    bleManager.stopDeviceScan();
                    device = bleDevice;

                    try {
                        await device.connect().then((connectedDevice) =>
                            connectedDevice.discoverAllServicesAndCharacteristics()
                        );
                        const services = await device.services();
                        const characteristics = await services[0].characteristics();
                        setCharacteristic(JSON.stringify(characteristics));
                        const array= []
                        let isStaticIpSupported = false;
                        for (const characteristic of characteristics) {
                            try {
                                const value = characteristic.value ? await base64.decode(characteristic.value).toString('utf8') : '';
                                array.push(value)
                                setHbdVersion(value.Hbd1);
                                if (value.includes("ethernet_static_ip")) {
                                    isStaticIpSupported = true;
                                    break;
                                }
                            } catch (err) {
                                console.warn(`Error decoding characteristic: ${err.message}`);
                            }
                        }
                        setCaps(array);

                        setIsSupportStaticIp(isStaticIpSupported);
                    } catch (err) {
                        setCapError(JSON.stringify(err));
                        return reject(new Error("Failed to connect or retrieve characteristics"));
                    } finally {
                        setLoading(false);
                        await bleManager.destroy();
                        resolve(device);
                    }
                }
            });

            setTimeout(async () => {
                await bleManager.stopDeviceScan();
                if (!device) reject(new Error("No device found within timeout period"));
            }, 10000);  // Adjust timeout duration as needed
        });
    }
```

### Relevant log output

```shell
IMG_2906.PNG
IMG_2907.PNG
IMG_2908.PNG
IMG_2909.PNG
IMG_2910.PNG
IMG_2911.PNG
IMG_2912.PNG
IMG_2913.PNG
IMG_2914.PNG
IMG_2915.PNG
IMG_2916.PNG
IMG_2917.PNG
IMG_2918.PNG
IMG_2919.PNG
IMG_2920.PNG
IMG_2921.PNG
IMG_2922.PNG
IMG_2923.PNG
IMG_2924.PNG
IMG_2925.PNG
IMG_2926.PNG
```

### Additional information

I am trying to decode from base64

RohitDeshwal, 2024-11-12:
i'm facing same issue

GBR-422777, 2024-11-17:
Any fix yet?

aliberski, 2025-01-18:
Thanks for reporting this! We’ve added it to our backlog and will review it soon.

dahlmo, 2025-01-23:
Confirming same behaviour on iOS 17.6.1

hasnatfahim12, 2025-01-28:
Facing similar issue trying to run _monitorCharacteristicForService_ api to monitor notification from devices, works properly with android but not with iOS (version 18.1)

hasnatfahim12, 2025-03-12:
Any fix yet?

dahlmo, 2025-03-14:
@hasnatfahim12 Maybe you can try [v3.5.0](https://github.com/dotintent/react-native-ble-plx/releases/tag/v3.5.0)? I see there's been some recent changes to the iOS characteristics part of the code, but it doesn't seem to be tracked so not completely sure if this will fix it.