dotintent/react-native-ble-plx #1249: 🐛 Not supporting Expo SDK 52

thinkbix, 2024-11-15:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

Getting in so many libraries, But sending any one

@expo/metro-config@0.19.4
node_modules/@expo/metro-config
  @expo/metro-config@"^0.19.4" from the root project
  @expo/metro-config@"~0.19.0" from @expo/cli@0.21.5
  node_modules/@expo/cli
    @expo/cli@"0.21.5" from expo@52.0.7
    node_modules/expo
      expo@"^52.0.7" from the root project
  @expo/metro-config@"0.19.4" from expo@52.0.7
  node_modules/expo
    expo@"^52.0.7" from the root project

@expo/metro-config@0.18.11
node_modules/react-native-ble-plx/node_modules/@expo/metro-config
  @expo/metro-config@"0.18.11" from @expo/cli@0.18.31
  node_modules/react-native-ble-plx/node_modules/@expo/cli
    @expo/cli@"0.18.31" from expo@51.0.39
    node_modules/react-native-ble-plx/node_modules/expo
      expo@"^51.0.14" from react-native-ble-plx@3.2.1
      node_modules/react-native-ble-plx
        react-native-ble-plx@"^3.2.1" from the root project
  @expo/metro-config@"0.18.11" from expo@51.0.39
  node_modules/react-native-ble-plx/node_modules/expo
    expo@"^51.0.14" from react-native-ble-plx@3.2.1
    node_modules/react-native-ble-plx
      react-native-ble-plx@"^3.2.1" from the root project

### Current Behavior

BLE Library Working but getting warnings and all

### Library version

3.2.1

### Device

iPhone 15 Pro, Android 8.0

### Environment info

```shell
expo-env-info 1.2.1 environment info:
    System:
      OS: macOS 12.7.6
      Shell: 5.8.1 - /bin/zsh
    Binaries:
      Node: 20.11.1 - ~/.nvm/versions/node/v20.11.1/bin/node
      Yarn: 1.22.22 - ~/.nvm/versions/node/v20.11.1/bin/yarn
      npm: 10.9.0 - ~/.nvm/versions/node/v20.11.1/bin/npm
      Watchman: 2024.08.26.00 - /usr/local/bin/watchman
    Managers:
      CocoaPods: 1.15.2 - /usr/local/bin/pod
    SDKs:
      iOS SDK:
        Platforms: DriverKit 22.2, iOS 16.2, macOS 13.1, tvOS 16.1, watchOS 9.1
      Android SDK:
        API Levels: 23, 25, 26, 29, 30, 31, 32, 33, 34
        Build Tools: 23.0.2, 25.0.2, 25.0.3, 26.0.1, 26.0.2, 27.0.0, 27.0.1, 27.0.3, 28.0.3, 29.0.2, 29.0.3, 30.0.2, 30.0.3, 32.0.0, 33.0.1, 34.0.0
        System Images: android-23 | Google APIs Intel x86 Atom, android-24 | Google Play Intel x86 Atom, android-30 | Google APIs Intel x86 Atom, android-30 | Google Play Intel x86 Atom, android-32 | Google APIs Intel x86 Atom_64
        Android NDK: 22.1.7171670
    IDEs:
      Android Studio: 2022.2 AI-222.4459.24.2221.10121639
      Xcode: 14.2/14C18 - /usr/bin/xcodebuild
    npmPackages:
      @expo/metro-config: ^0.19.4 => 0.19.4 
      expo: ^52.0.0 => 52.0.6 
      expo-router: ~4.0.5 => 4.0.5 
      react: 18.3.1 => 18.3.1 
      react-dom: 18.3.1 => 18.3.1 
      react-native: ^0.76.1 => 0.76.1 
    npmGlobalPackages:
      eas-cli: 13.2.1
      expo-cli: 6.3.12
    Expo Workflow: managed
```

### Steps to reproduce

1. …
2. …

### Formatted code sample or link to a repository

```js
// Start Scanning
    const bleManager = new BleManager();

    // Bluetooth Permission
    const requestBluetoothPermissions = async (): Promise<boolean> => {
        if (Platform.OS === 'android') {
            // const granted = await PermissionsAndroid.request(
            //     PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION
            // );
            // return granted === PermissionsAndroid.RESULTS.GRANTED;
            try {
                const permissions = [
                    PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
                    PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN,
                    PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT,
                ];

                const granted = await PermissionsAndroid.requestMultiple(permissions);

                if (
                    granted['android.permission.ACCESS_FINE_LOCATION'] === PermissionsAndroid.RESULTS.GRANTED &&
                    granted['android.permission.BLUETOOTH_SCAN'] === PermissionsAndroid.RESULTS.GRANTED &&
                    granted['android.permission.BLUETOOTH_CONNECT'] === PermissionsAndroid.RESULTS.GRANTED
                ) {
                    return true;
                } else {
                    Alert.alert('Permissions Denied', 'Bluetooth permissions are required to connect to devices.');
                    return false;
                }
            } catch (error) {
                Alert.alert('Error', 'Failed to request permissions');
                return false;
            }
        }
        return true;
    }

    // Handle Scan Printers
    const handleScanPrinters = async () => {
        const permissionsGranted = await requestBluetoothPermissions();
        if (!permissionsGranted) return;

        // Set Flag
        setScanning(true);
        handleHapticFeedback();

        try {
            bleManager.startDeviceScan(null, null, (error: BleError | null, device: Device | null) => {
                if (error && error?.reason) {
                    Alert.alert(`${error?.errorCode}`, `${error.reason}`);
                    setScanning(false);
                    return;
                }
                // Only proceed if device has both id and name
                if (device && device.name && device.id) {
                    setPrinters((prevList) => {
                        const deviceExists = prevList.some((d) => d.id === device.id);
                        if (!deviceExists) {
                            return [...prevList, { id: device.id, name: device.name }];
                        }
                        return prevList;
                    });
                }
            });
            setTimeout(() => {
                bleManager.stopDeviceScan();
                setScanning(false);
            }, 15000);
        } catch (error: any) {
            Alert.alert('Error!', `${error.message}`);
            setScanning(false);
        } finally {
            // bleManager.destroy();
            // setScanning(false);
        }
    }

    // Handle Printer
    const handlePrinter = async (id: string) => {
        setPrinterId(id);
        handleHapticFeedback();
        return;
    }
```

### Relevant log output

```shell
Getting issues in below libraries just because of BLE library

The package "expo-modules-autolinking" should not be installed directly in your project. It is a dependency of other Expo packages, which will install it automatically as needed.

Expected package expo-modules-autolinking@~2.0.0
Found invalid:
  expo-modules-autolinking@1.11.3
  (for more info, run: npm why expo-modules-autolinking)
Expected package @expo/config-plugins@~9.0.0
Found invalid:
  @expo/config-plugins@8.0.11
  (for more info, run: npm why @expo/config-plugins)
Expected package @expo/prebuild-config@~8.0.0
Found invalid:
  @expo/prebuild-config@7.0.9
  (for more info, run: npm why @expo/prebuild-config)
Expected package @expo/metro-config@~0.19.0
Found invalid:
  @expo/metro-config@0.18.11
  (for more info, run: npm why @expo/metro-config)
```

akhockey21, 2024-11-15:
I've added a PR:
https://github.com/dotintent/react-native-ble-plx/pull/1250

thinkbix, 2024-11-19:
> I've added a PR: #1250

Is it solved? If yes then please guide me what i have to do at my end?

akhockey21, 2024-11-22:
> > I've added a PR: #1250
> 
> Is it solved? If yes then please guide me what i have to do at my end?

Until the PR is merged, feel free to just update your package.json dep with this:
`"react-native-ble-plx": "github:akhockey21/react-native-ble-plx",`

g-otn, 2024-11-26:
Shouldn't `expo` be a peerDependency anyways?

g-otn, 2024-11-26:
As a workaround, you may update `react-native-ble-plx` dependencies using npm overrides (or yarn resolutions, etc):

```json5
// package.json
{
  // ...
  "overrides": {
    "react-native-ble-plx": {
      "expo": "~52.0.11",
      "@expo/config-plugins": "^9.0.0"
    }
   // ...
  }
}

aliberski, 2025-01-08:
This should be solved in the latest version 3.4.0

closed 2025-01-08 by aliberski