dotintent/react-native-ble-plx #1255: Question - Reading SSID list from Raspberry Device

pzajas, 2024-12-10:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

Hello!

I'm building an app that connects to raspberry device via ble to pass wifi credentials to that app. It's working now but the SSID json? I recive seems fairly malformed. I'm able to get one clean ssid only from it but I should get like 10+. It looks like this after some parsing:

 (NOBRIDGE) LOG  [DEBUG] Extracted 41 SSIDs before cleaning: ["CGA2121_QV5rJnx", "2.", ",", "CGA2121_QV5rJnx", "2.", ",", "CGA2121_QV5rJnx", "2.", ",", "CGA2121_QV5rJnx", "2.", ",", "CGA2121_QV5rJnx", "2.", ",", "CGA2121_QV5rJnx", "2.", ",", "CGA2121_QV5rJnx", "2.", ",", 
"CGA2121_QV5rJnx", "2.", ",", "CGA2121_QV5rJnx", "2.", ",", "CGA2121_QV5rJnx", "2.", ",", "CGA2121_QV5rJnx", "2.", ",", "CGA2121_QV5rJnx", "2.", ",", "CGA2121_QV5rJnx", "2.", ",", "CGA2121_QV5rJnx", "2."]

This is raw:

WyJDR0EyMTIxX1FWNXJKbngiLCIyLlsiQ0dBMjEyMV9RVjVySm54IiwiMi5bIkNHQTIxMjFfUVY1ckpueCIsIjIuWyJDR0EyMTIxX1FWNXJKbngiLCIyLlsiQ0dBMjEyMV9RVjVySm54IiwiMi5bIkNHQTIxMjFfUVY1ckpueCIsIjIuWyJDR0EyMTIxX1FWNXJKbngiLCIyLlsiQ0dBMjEyMV9RVjVySm54IiwiMi5bIkNHQTIxMjFfUVY1ckpueCIsIjIuWyJDR0EyMTIxX1FWNXJKbngiLCIyLlsiQ0dBMjEyMV9RVjVySm54IiwiMi5bIkNHQTIxMjFfUVY1ckpueCIsIjIuWyJDR0EyMTIxX1FWNXJKbngiLCIyLlsiQ0dBMjEyMV9RVjVySm54IiwiMi5bIkNHQTIxMjFfUVY1ckpueCIsIjIuWyJDR0EyMTIxX1FWNXJKbngiLCIyLlsiQ0dBMjEyMV9RVjVySm54IiwiMi5bIkNHQTIxMjFfUVY1ckpueCIsIjIuWyJDR0EyMTIxX1FWNXJKbngiLCIyLlsiQ0dBMjEyMV9RVjVySm54IiwiMi5bIkNHQTIxMjFfUVY1ckpueCIsIjIuWyJDR0EyMTIxX1FWNXJKbngiLCIyLlsiQ0dBMjEyMV9RVjVySm54IiwiMi5bIkNHQTIxMjFfUVY1ckpueCIsIjIuWyJDR0EyMTIxX1FWNXJKbngiLCIyLlsiQ0dBMjEyMV9RVjVySm54IiwiMi5bIkNHQTIxMjFfUVY1ckpueCIsIjIuWyJDR0Ey

Also, I have React.js app that does the same thing and everything works correctly there:

['CGA2121_QV5rJnx', '2.4G-Vectra-WiFi-B591DC', '2.4G-Vectra-WiFi-9DC6A4', '5G-Vectra-WiFi-B591DC', 'UPC1717935', '5G-Vectra-WiFi-9DC6A4', 'CGA2121_5b3dqhu', 'DCF979', 'TP-LINK_87D7C6', 'UPCB8BA3BF', 'UPC7650130', 'KODPRIVa171']

Any ideas about what is wrong in my case? Would be grateful for each hint guys.

Tanks in advance! :D

### Question related code

```markdown
const getSSIDList = async (device: Device, serviceUUID: string, characteristicUUID: string) => {
    try {
      const characteristic = await device.readCharacteristicForService(serviceUUID, characteristicUUID)
      const decodedBase64Value = base64.decode(characteristic.value)
      const buffer = stringToArrayBuffer(decodedBase64Value)
      const decodedWithAB2Str = ab2str(buffer)
      const ssidList = decodedWithAB2Str.match(/"([^"]+)"/g).map((ssid) => ssid.replace(/["\[\]]/g, '').trim())
      const cleanedSSIDs = ssidList.filter((ssid) => ssid && ssid.length > 4 && !ssid.includes('2.['))
      const uniqueSSIDs = Array.from(new Set(cleanedSSIDs))
      await AsyncStorage.setItem('ssids', JSON.stringify(uniqueSSIDs))

      return uniqueSSIDs
    } catch (error) {
      console.error('Error retrieving SSID list from device:', error)
      return []
    }
  }

  const stringToArrayBuffer = (str: string): ArrayBuffer => {
    const encoder = new TextEncoder()
    return encoder.encode(str).buffer
  }
```

pzajas, 2025-01-13:
> @pzajas could you confirm the platform you are testing on? (ios/android) What version of the library are you using?

It is on android currently. I use the version from custom fix branch:

github:akhockey21/react-native-ble-plx#Expo-52-Support

The thing is the ssid list is in base64 string format. It works well on the web app but on mobile the string is malformed, dunno why. The backend  team changed the forat of ssid list to simple json and it is fine now I guess.

aliberski, 2025-01-16:
@pzajas could you share raw string before decoding? I mean `characteristic.value`. Maybe it's something wrong with decoding method you're using

aliberski, 2025-01-16:
btw @pzajas you can use the latest version `3.4.0` instead of the custom fix branch. We removed expo dependency so it should work with the latest expo

pzajas, 2025-01-16:
The raw string is in the first comment,

WyJDR0EyMTIxX1FWNXJKbngiLCIyLlsiQ0dBMjEyMV9RVjVySm54IiwiMi5bIkNHQTIxMjFfUVY1ckpueCIsIjIuWyJDR0EyMTIxX1FWNXJKbngiLCIyLlsiQ0dBMjEyMV9RVjVySm54IiwiMi5bIkNHQTIxMjFfUVY1ckpueCIsIjIuWyJDR0EyMTIxX1FWNXJKbngiLCIyLlsiQ0dBMjEyMV9RVjVySm54IiwiMi5bIkNHQTIxMjFfUVY1ckpueCIsIjIuWyJDR0EyMTIxX1FWNXJKbngiLCIyLlsiQ0dBMjEyMV9RVjVySm54IiwiMi5bIkNHQTIxMjFfUVY1ckpueCIsIjIuWyJDR0EyMTIxX1FWNXJKbngiLCIyLlsiQ0dBMjEyMV9RVjVySm54IiwiMi5bIkNHQTIxMjFfUVY1ckpueCIsIjIuWyJDR0EyMTIxX1FWNXJKbngiLCIyLlsiQ0dBMjEyMV9RVjVySm54IiwiMi5bIkNHQTIxMjFfUVY1ckpueCIsIjIuWyJDR0EyMTIxX1FWNXJKbngiLCIyLlsiQ0dBMjEyMV9RVjVySm54IiwiMi5bIkNHQTIxMjFfUVY1ckpueCIsIjIuWyJDR0EyMTIxX1FWNXJKbngiLCIyLlsiQ0dBMjEyMV9RVjVySm54IiwiMi5bIkNHQTIxMjFfUVY1ckpueCIsIjIuWyJDR0EyMTIxX1FWNXJKbngiLCIyLlsiQ0dBMjEyMV9RVjVySm54IiwiMi5bIkNHQTIxMjFfUVY1ckpueCIsIjIuWyJDR0Ey

Well, it is clearly malformed and that's the one i get from the raspberry. Don't know why - maybe I have so many ssids in the vicinity that the base64 format cannot handle it properly. I tried many methods to decode it, even tried external websites.

aliberski, 2025-01-21:
@pzajas I tested this case (on android) and for my device it returns correct characteristic that I can decode with base64. I believe the error is not related to the library, but to the bluetooth device you are scanning (so it returns the malformed data).
I'm closing the issue for now. Let me know if you had any other questions.

closed 2025-01-21 by aliberski

pzajas, 2025-01-22:
> [@pzajas](https://github.com/pzajas) I tested this case (on android) and for my device it returns correct characteristic that I can decode with base64. I believe the error is not related to the [...]

Well, I thought about this one too. We have workaround anyway but I wanted to check if u know/met the issue. Happy coding! :)