dotintent/react-native-ble-plx #1259: Support bluetoothAlwaysPermission localization via config plugin

JakeYallop, 2024-12-13:
### Feature Description

Expo supports localization of app metadata: https://docs.expo.dev/guides/localization/#translating-app-metadata

This functionality is not surfaced in the config plugin options.

### Current Workaround

Configure the app configuration separately to the `react-native-ble-plx` plugin. This can be ambigious, as if I have a `bluetoothAlwaysPermission` string specified, in addition to localized settings configured elsewhere, its not clear precisely what the behaviour would be.

### Additional Information

- [X] I have searched for existing feature requests.
- [ ] I would like to work on this feature myself.

aliberski, 2025-01-09:
@JakeYallop this is how the code of setting `NSBluetoothAlwaysUsageDescription` in the plugin look like

```
const BLUETOOTH_ALWAYS = 'Allow $(PRODUCT_NAME) to connect to bluetooth devices'

config.modResults.NSBluetoothAlwaysUsageDescription =
      bluetoothAlwaysPermission || config.modResults.NSBluetoothAlwaysUsageDescription || BLUETOOTH_ALWAYS
```

It means you can simply ignore passing `bluetoothAlwaysPermission` parameter to the plugin and your `Info.plist` values will be used. 
For example:
```
// app.json

{
   "expo":{
      "plugins":[
         [
            "react-native-ble-plx",
            {
               "isBackgroundEnabled":true,
               "modes":[
                  "peripheral",
                  "central"
               ]
            }
         ]
      ],
      "ios":{
         "infoPlist":{
            "CFBundleAllowMixedLocalizations":true
         }
      },
      "locales":{
         "en":"./en.json",
         "de":"./de.json"
      }
   }
}
```

```
// en.json

{
  "NSBluetoothAlwaysUsageDescription": "We need Bluetooth to connect to devices."
}
```

```
// de.json

{
  "NSBluetoothAlwaysUsageDescription": "Wir benötigen Bluetooth, um eine Verbindung zu Geräten herzustellen."
}
```

closed 2025-01-09 by aliberski

aliberski, 2025-01-09:
I added this example to our wiki [HERE](https://github.com/dotintent/react-native-ble-plx/wiki/Expo#localize-nsbluetoothalwaysusagedescription-in-ios)