dotintent/react-native-ble-plx #1261: How to disconnect the devices

MAYILERUMPERUMAL, 2024-12-20:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

i am connected the devices and read the devices value but how to disconnect the device using cancelDeviceConnection
does not working

benevolentsocialplanner, 2024-12-29:
```
const disconnectFromDevice = () => {
        if (connectedDevice) {
            bleManager.cancelDeviceConnection(connectedDevice.id);
            setConnectedDevice(null);
        }
    };
```

the connectedDevice object here is what you get from: 

```
const deviceConnection = await bleManager.connectToDevice(device.id);
await deviceConnection.discoverAllServicesAndCharacteristics();
setConnectedDevice(deviceConnection);
```

aliberski, 2025-01-08:
@MAYILERUMPERUMAL this should work as @benevolentsocialplanner described. You should pass connected device id to `cancelDeviceConnection`. I hope it answers your question.

closed 2025-01-08 by aliberski