dotintent/react-native-ble-plx #1263: How to delete the bonding handled on OS level?

benevolentsocialplanner, 2024-12-29:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

After i connect the bonding is handling on the operating system side. Since they are bonded the ESP32 device will always return the same device ID. However, in the docs i couldn't find a way to manually delete the bond. I store the connected device in the storage. I can delete it from storage but that will leave the bonding open. 

Thanks in advance.

benevolentsocialplanner, 2025-01-06:
Still valid

closed 2025-01-16 by aliberski

olalonde, 2025-04-09:
Would you consider a PR adding this feature?

benevolentsocialplanner, 2025-04-09:
> Would you consider a PR adding this feature?

Idk how to fix that but will take a look if i can come up with a solution.