dotintent/react-native-ble-plx #1264: IOS scanning in background mode only starts working when flipping BT on/off in settings.

niclaslindberg, 2025-01-06:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

I'm running my app (based on Expo SDK 51) in the background. Scanning is started to searching for specific serviceUUID. Everything works perfectly in foreground. My problem is in the background scanning mode.
The device I want to connect to is broadcasting but the app doesn't connect. However, if I enter IOS settings and toggle BT on/off (without putting my app in the foreground) while the app is in the background the app connects to the device correctly.  
I also noticed that starting the NRF Scanner App scan function triggers my app to connect to the device the same way as IOS settings BT on/off (the app being in the background all the time). 

### Question related code

```markdown
// bleutooth-manager-factory.ts

import { BleError, BleErrorCode, BleManager, BleRestoredState, State } from "react-native-ble-plx";
import { createLogger } from "../../logging";

const bluetoothStartTimeout = 5000

const logger = createLogger("ble-manager-factory")

const createBleManager = ():BleManager => {
    return new BleManager({
        restoreStateIdentifier: "some-uid",
        restoreStateFunction: (restoredState) => {
            if (restoredState) {
                logger.debug("Restored BLE state:", restoredState);
    
                restoredState.connectedPeripherals.forEach(async (device) => {
                    logger.debug("Reconnecting to:", device.name);
                    try {
                        const connectedDevice = await device.connect();
                        logger.debug("Successfully reconnected to:", connectedDevice.name);
    
                        await connectedDevice.discoverAllServicesAndCharacteristics();
                    } catch (error) {
                        logger.error("Error reconnecting to device:", error);
                    }
                });
            } else {
               logger.debug("No BLE state to restore. App launched fresh.");
            }
        }
    })
}

export class BluetoothManagerFactory {

    private bleManager:BleManager|undefined 

    async getManager():Promise<BleManager> {
        const bleManager = this.bleManager = this.bleManager ?? createBleManager()
        try {
            const state = await bleManager.state()
            if(state === State.PoweredOn) {
                return this.bleManager
            }
        } catch (e) {
            if(e instanceof BleError && e.errorCode===BleErrorCode.BluetoothManagerDestroyed) {
                logger.debug("Re-creating ble-manager. Destroyed.")
                this.bleManager = undefined
            } else {
                throw e
            }
        }
        return new Promise<BleManager>((resolve,reject) => {
            const bleManager = this.bleManager = this.bleManager ?? createBleManager()
            const timeout = setTimeout(() => reject(new Error(`Could not start bluetooth native service.`)),bluetoothStartTimeout)
            bleManager.onStateChange((state) => {
                if (state === State.PoweredOn) {
                    clearTimeout(timeout)
                    resolve(bleManager)
                }
    
            }, true)
        })
    }
}
    

export const BluetoothManagerFactoryInstance = new BluetoothManagerFactory()

App uses BluetoothManagerFactoryInstance.getManager() to get hold of a valid BLEManager.
```

niclaslindberg, 2025-01-07:
I've added the same question to SO: https://stackoverflow.com/questions/79337172/ios-react-native-ble-plx-background-scanning-works-only-after-toggling-bluetooth

andrewcharnley, 2025-01-07:
I just came here to report another bug and it's very similar. On iOS, connection works fine but after turning bluetooth off and then back on startDeviceScan wont pick anything up and it gets stuck.

This behaviour is different from Android where it works and I also verified iOS wasn't somehow hanging onto the connection as an Android phone can now connect to the peripheral. 

Quitting the App resets and it connects again. I also tried .destroy() with no luck.

niclaslindberg, 2025-01-07:
Interresting. That’s kind of the opposite problem. My issue is that I have already started deviceScan while in foreground. Move the app to background. Nothing happens until toggling BT on/off then it connects while still in background. Thanks for pointing out! Let’s elaborate!

niclaslindberg, 2025-01-19:
Ok. Problem solved. It actually turned out that the problem was in the firmware of my device (which I also control). The firmware was advertising serviceUID in "scanResponse" message and not in the primary advertising package from the device. It turns out that IOS doesn't ask for "additional scanResponse" in background mode hence no detected advertising.
Solution: Change device implementation to include serviceUID in primary advertising message and not in the scanResponse message.

closed 2025-01-21 by aliberski

ducpt-bili, 2025-04-10:
hey @niclaslindberg , i'm trying to make ios work on the background mode. In my case, my app connect iot devices then go to background, i want to send data from iot devices near realtime as much as possible. May you sent me the sample code or your implemetation? any link to code for help.
Thanks.

niclaslindberg, 2025-04-10:
Hi @ducpt-bili 
What is the problem? 🤕
Connecting, transmitting data?

ducpt-bili, 2025-04-10:
hi @niclaslindberg , both connecting and transfer data in background.
Thank you for your response.

niclaslindberg, 2025-04-10:
Ok. My learnings:
1. You can have the same BleManager instance running through out the whole application lifecycle. No need to restart/re-instantiate ever.
2. You MUST scan for primary serviceUUIDs of your device. The primary serviceUUID MUST be a part of your advertising package. Make sure (this is hard with tools like NRF connect etc) that the primary serviceUUID is not part of secondary advertsing package (ServerResponse). Server response messages are not asked for by IOS when app is in background.
3. For IOS to reconnect with a device when in background YOU MUST trigger a connection attempt when the device drops the connection. Otherwise advertising packages when IOS is in background will not surface to the app.

Besides all the above transmitting data is the same as in foreground; subscribing to you read-characteristic and writing to your write-characteristic. Sorry, have no code to share.

ducpt-bili, 2025-04-10:
hi @niclaslindberg , thanks for sharing.