dotintent/react-native-ble-plx #1266: Feature Request:

kostas1, 2025-01-15:
### Feature Description

Hey,

I have a characteristic that is both isIndicatable=true and isNotifiable=true and I need to subscribe to indications rather than notifications.

The issue is that code is favoring notifications, as suggested on BleManager.monitorCharacteristicForDevice and also within BleModule.safeMonitorCharacteristicForDevice inside Android codebase.

Thank you

### Current Workaround

No workaround, it's a blocker for my app

### Suggested Implementation

Can you please add a hint into monitorCharacteristicForDevice, perhaps like:

monitorCharacteristicForDevice(
      deviceIdentifier: DeviceId,
      serviceUUID: UUID,
      characteristicUUID: UUID,
      listener: (error: BleError | null, characteristic: Characteristic | null) => void,
      transactionId?: TransactionId,
      subscriptionType?: "Notification" | "Indication" = "Notification",
    ): Subscription

so it's backwards compatible, but allows for more flexibility.

And native code paths could check on the value rather than preferring notifications

### Additional Information

- [X] I have searched for existing feature requests.
- [ ] I would like to work on this feature myself.

aliberski, 2025-01-23:
@kostas1 thank you for creating the issue. I will work on the enhancement. Keep in mind this will work only on android. Note from our wiki:
```
On vanilla Android BLE API it is possible to choose wether notification or indication is setup if a characteristic supports both but the iOS does not allow to choose—that is why it is not possible to do so in this library
```

closed 2025-01-24 by aliberski

kostas1, 2025-01-27:
oh wow thank you! do you have any plans to bump up a release version anytime soon and make it available through npm/expo? :)

aliberski, 2025-02-03:
@kostas1 we plan to push a new release this week

PR #1269 (2025-01-24): feat: add subscriptionType param to monitor characteristic methods
closes #1266