dotintent/react-native-ble-plx #1267: Enable bluetooth method is deprecated in Android API 33

marllonfrizzo, 2025-01-17:
Hello!

After updating from version 3.2.1 to version 3.4.0, the `enable` method stopped working on Android API 33.

I noticed that this function was deprecated on Android API 33 and to activate Bluetooth programmatically, it is necessary to request user permission.

According to the [documentation](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#enable()), for this request to work, an `ACTION_REQUEST_ENABLE` Intent must be used. I have no experience with Android development, but checking the changes that occurred between versions 3.2.1 and 3.4.0, this PR https://github.com/dotintent/react-native-ble-plx/pull/1232 possibly introduced the change that caused the reported problem as a side effect. By reverting this change for testing, the enable method works by opening a dialog for the user to provide this permission.

I believe that reverting is not a solution, since this change fixes other problems, but perhaps think of some other solution to work on Android API 33 or update the documentation regarding this.

Best regards!

Rafaelbg107, 2025-01-17:
Hi! 

At the meantime I found a workaround using [expo-intent-launcher](https://docs.expo.dev/versions/latest/sdk/intent-launcher/) to access `ACTION_REQUEST_ENABLE`.

Your enabler function should be like

```
const enableBT = async () => {
  if (Platform.OS === 'android') {
    try {
      startActivityAsync('android.bluetooth.adapter.action.REQUEST_ENABLE');
    } catch (error) {
      console.log(error)
    }
  }
}
```
At least while it is fixed.

aliberski, 2025-01-18:
Thank you for bringing this up! We've added the issue to our backlog and will work on a fix soon. Contributions are always welcome if you'd like to help out!

marllonfrizzo, 2025-01-21:
> Hi!
>
> At the meantime I found a workaround using [expo-intent-launcher](https://docs.expo.dev/versions/latest/sdk/intent-launcher/) to access `ACTION_REQUEST_ENABLE`.
>
> Your enabler function should be [...]

Hello,

Thanks for the tip, it worked!

Here I ended up using react-native's own Linking, it worked too.

It looked like this
```ts
Linking
  .sendIntent('android.bluetooth.adapter.action.REQUEST_ENABLE')
  .catch(error => {
    console.warn(error);
    Linking.openSettings();
  });
```
> Thank you for bringing this up! We've added the issue to our backlog and will work on a fix soon. Contributions are always welcome if you'd like to help out!

Thanks for the quick response!

I have no experience with Android development, but I'll give it a try.

Best regards!

Rafaelbg107, 2025-01-21:
> Thank you for bringing this up! We've added the issue to our backlog and will work on a fix soon. Contributions are always welcome if you'd like to help out!

I've never submitted a fix on a public project but I'll give it a shot!