dotintent/react-native-ble-plx #1275: BLE Devices on iOS 18.3 do not show their UUIDs

micheleScagliola00, 2025-02-07:
I'm developing an application that needs to use BLE devices' UUIDs to recover data from an API, but the scanned device in my code shows as id parameter one that doesn't correspond to the one I set with other apps. Moreover this id varies depending on which device I use for scanning. I ask if there is a way to recover the UUID using this library.

jerna-sn0482828, 2025-02-07:
Hello, we tried to solve the issue.

This is what we did:

Modified BLEService to extract UUID from advertising data and updated DashboardScreen to display the UUID alongside other device information.

You can review changes in this commit: https://github.com/jerna-sn0482828/dotintent-react-native-ble-plx-1275/commit/e0c95fa1aa26f28a41e15a56b46f4f5079aad453.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

Latta AI seeks to solve problems in open source projects as part of its mission to support developers around the world. Learn more about our mission at https://latta.ai/ourmission . If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

aliberski, 2025-05-26:
Hello @micheleScagliola00. Is this issue specific for iOS of 18.3 and works on other versions? What devices are you scanning?
Please provide more information if the issue is still bothers you.

closed 2025-05-26 by aliberski