dotintent/react-native-ble-plx #1276: 🐛 writecharacteristicwithoutresponsefordevice in iOS have issue

tuanLG1995, 2025-02-09:
### Prerequisites

- [x] I checked the documentation and FAQ without finding a solution
- [x] I checked to make sure that this issue has not already been filed

### Expected Behavior

Dear Admin, 
I try to use function writecharacteristicwithoutresponsefordevice in version Android is working well 
But In version iOS, I can't send data
Could you help me check about that? 
I share the information: 
"react-native-ble-plx": "^3.4.0",
"react-native": "0.68.2",
"react": "17.0.2",
 BLTManager.writeCharacteristicWithoutResponseForDevice(
          connectStatusRef.current!!.id,
          BLUETOOTH_UUID,
          BLUETOOTH_CHARACTERISTIC_UUID,
          converted.toString('base64'),
        );

### Current Behavior

BLTManager.writeCharacteristicWithoutResponseForDevice(
          connectStatusRef.current!!.id,
          BLUETOOTH_UUID,
          BLUETOOTH_CHARACTERISTIC_UUID,
          converted.toString('base64'),
        );

### Library version

3.4.0

### Device

iPhone 11PRM (17.4.1)

### Environment info

```shell
"react-native-ble-plx": "^3.4.0",
"react-native": "0.68.2",
"react": "17.0.2",
```

### Steps to reproduce

Send data use function writeCharacteristicWithoutResponseForDevice

### Formatted code sample or link to a repository

```js
send message use function writeCharacteristicWithoutResponseForDevice
```

### Relevant log output

```shell
N/A
```

tuanLG1995, 2025-02-09:
I add the log 
2025-02-09 21:17:22.988430+0700 StemApp[486:14493] [javascript] useEffect 1
2025-02-09 21:17:23.192815+0700 StemApp[486:14493] [javascript] Connection established
2025-02-09 21:17:23.198391+0700 StemApp[486:14493] [javascript] clearInterval
2025-02-09 21:17:23.198413+0700 StemApp[486:14485] [CoreBluetooth] WARNING: Characteristic <CBCharacteristic: 0x281356ac0, UUID = BEB5483E-36E1-4688-B7F5-EA07361B26A8, properties = 0xA, value = (null), notifying = NO> does not specify the "Write Without Response" property - ignoring response-less write
2025-02-09 21:17:23.204191+0700 StemApp[486:14493] [javascript] { object: 'read', type: 'BATTERY', message: '' }
2025-02-09 21:17:23.204305+0700 StemApp[486:14493] [javascript] 'Read Message -> ', 'BATTERY', ''
2025-02-09 21:17:23.204599+0700 StemApp[486:14493] [javascript] 'heightBuffer  --> ', { '0': 255, '1': 85, '2': 3, '3': 0, '4': 1, '5': 60 }
2025-02-09 21:17:23.205587+0700 StemApp[486:14493] [javascript] useEffect 1
2025-02-09 21:17:23.377829+0700 StemApp[486:14493] [javascript] Possible Unhandled Promise Rejection (id: 0):
TypeError: undefined is not a function
TypeError: undefined is not a function
    at _callee7$ (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:131368:119)
    at call (native)
    at tryCatch (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:19861:23)
    at anonymous (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:19942:32)
    at anonymous (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:19886:30)
    at call (native)
    at tryCatch (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:19861:23)
    at invoke (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:19892:30)
    at anonymous (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:19913:21)
    at tryCallTwo (/Users/distiller/hermes/build_iphoneos/lib/InternalBytecode/InternalBytecode.js:61:9)
    at doResolve (/Users/distiller/hermes/build_iphoneos/lib/InternalBytecode/InternalBytecode.js:216:25)
    at Promise (/Users/distiller/hermes/build_iphoneos/lib/InternalBytecode/InternalBytecode.js:82:14)
    at callInvokeWithMethodAndArg (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:19912:35)
    at value (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:19916:159)
    at anonymous (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:19886:30)
    at anonymous (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:20022:69)
    at _callee7 (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:131364:40)
    at anonymous (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:131395:25)
    at commitHookEffectListMount (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:12716:38)
    at commitPassiveMountOnFiber (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:13704:44)
    at commitPassiveMountEffects_complete (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:13676:40)
    at commitPassiveMountEffects_begin (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:13666:47)
    at commitPassiveMountEffects (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:13656:40)
    at flushPassiveEffectsImpl (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:14745:34)
    at flushPassiveEffects (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:14709:43)
    at performSyncWorkOnRoot (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:14231:28)
    at flushSyncCallbacks (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:6194:36)
    at flushSyncCallbacksOnlyInLegacyMode (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:6179:29)
    at scheduleUpdateOnFiber (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:13989:49)
[... cut ...]
    at __guard (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:2527:15)
    at flushedQueue (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:2354:21)
    at invokeCallbackAndReturnFlushedQueue (http://192.168.1.2:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=com.vrbox.vietnam:2348:33)
2025-02-09 21:17:23.443269+0700 StemApp[486:14493] [javascript] 'Converted receiving', { deviceID: 60, value: 10 }

tuanLG1995, 2025-02-13:
No have any support to me 👎

aliberski, 2025-05-26:
@tuanLG1995 which one is the output of `writeCharacteristicWithoutResponseForDevice` in the log you sent?