dotintent/react-native-ble-plx #1277: Question

gsagergithub, 2025-02-20:
### Prerequisites

- [x] I checked the documentation and FAQ without finding a solution
- [x] I checked to make sure that this issue has not already been filed
- [x] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [x] I'm running the latest version

### Question

Hi, guys. I was wondering if we're gonna have support for the new architecture. I updated to RN 0.76.7 recently and I found that the app is crashing on BLE connection. Problem I was able to solve after disabling the new architecture from gradle.properties. Ty very much.

### Question related code

```markdown

```

aliberski, 2025-02-21:
Hello @gsagergithub. Are you using the latest version of the library?

gsagergithub, 2025-02-21:
> Hello [@gsagergithub](https://github.com/gsagergithub). Are you using the latest version of the library?

Yes, I'm using the last version. 3.5.0. Ty

sebirdman, 2025-02-21:
I'm seeing a similar issue on react native 0.77.1 here are my findings so far:

1) it only seems to happen in release mode, not development mode.
2) I can only reproduce on iOS.
3) the issue is pretty annoying to track down, as it's almost certainly happening in the native side of things. I had to go library by library to figure out which native thing was doing it. 
4) as soon as i do `new BLEManager`, the crash happens - prior to the connection for me.

gsagergithub, 2025-02-21:
I'll share native logs soon

gsagergithub, 2025-02-22:
Guys, you can disregard my case. The issue was resolved by updating react-native-screens, @react-navigation/material-top-tabs, @react-navigation/, and @react-navigation/native-stack.  

Thanks, and sorry for the inconvenience.

mario688, 2025-02-24:
> I'm seeing a similar issue on react native 0.77.1 here are my findings so far:
> 
> 1. it only seems to happen in release mode, not development mode.
> 2. I can only reproduce on iOS.
> 3. the issue is pretty annoying to track down, as it's almost certainly happening in the native side of things. I had to go library by library to figure out which native thing was doing it.
> 4. as soon as i do `new BLEManager`, the crash happens - prior to the connection for me.
> 
> For now, i've disabled the new architecture on iOS, but that's not ideal.

@sebirdman  I've created a patch for this issue: https://github.com/dotintent/react-native-ble-plx/issues/1278