dotintent/react-native-ble-plx #1278: [iOS][New Architecture][Release] App crash  -- TEMPORARY FIX

mario688, 2025-02-24:
Hi,
In iOS release mode under the new react-native architecture, the app crashes. I've created a temporary patch for this issue.

My versions:
 "react-native": "0.77.1",
 "react-native-ble-plx": "^3.5.0",

**react-native-ble-plx+3.5.0.patch**

```
diff --git a/node_modules/react-native-ble-plx/ios/BlePlx.m b/node_modules/react-native-ble-plx/ios/BlePlx.m
index 8e0da36..2e28852 100644
--- a/node_modules/react-native-ble-plx/ios/BlePlx.m
+++ b/node_modules/react-native-ble-plx/ios/BlePlx.m
@@ -54,10 +54,14 @@ + (BOOL)requiresMainQueueSetup {
     return YES;
 }
 
-RCT_EXPORT_METHOD(createClient:(NSString*)restoreIdentifierKey) {
-    _manager = [BleAdapterFactory getNewAdapterWithQueue:self.methodQueue
+RCT_EXPORT_METHOD(createClient:(id)restoreIdentifierKey) {
+  if (restoreIdentifierKey == nil || [restoreIdentifierKey isEqual:[NSNull null]] ||
+      ([restoreIdentifierKey isKindOfClass:[NSString class]] && [(NSString *)restoreIdentifierKey length] == 0)) {
+    restoreIdentifierKey = nil;
+  }
+  _manager = [BleAdapterFactory getNewAdapterWithQueue:dispatch_get_main_queue()
                                     restoreIdentifierKey:restoreIdentifierKey];
-    _manager.delegate = self;
+  _manager.delegate = self;
 }
 
 RCT_EXPORT_METHOD(destroyClient) {
@@ -97,7 +101,13 @@ - (void)invalidate {
 
 RCT_EXPORT_METHOD(startDeviceScan:(NSArray*)filteredUUIDs
                           options:(NSDictionary*)options) {
-    [_manager startDeviceScan:filteredUUIDs options:options];
+  if (filteredUUIDs == nil || [filteredUUIDs isEqual:[NSNull null]]) {
+    filteredUUIDs = @[];
+  }
+  if (options == nil || [options isEqual:[NSNull null]]) {
+    options = @{};
+  }
+  [_manager startDeviceScan:filteredUUIDs options:options];
 }
 
 RCT_EXPORT_METHOD(stopDeviceScan) {
```

g-otn, 2025-03-10:
What does this patch do? Is it better than disabling new arch for iOS?

craiganderson-iotv, 2025-03-17:
Thanks @mario688 this is terrific!

Could it be related to this change in React Native `0.77.1`?
> Interop Layer: Properly handle null values coming from NativeModules. ([475f797a51](https://github.com/facebook/react-native/commit/475f797a51237e63e4d2bb4e5718e6aa73157767) by [sammy-SC](https://github.com/sammy-SC))

@g-otn It looks like it does some basic `null`-checking of `restoreIdentifierKey` and `filteredUUIDs` when starting and scanning, respectively. Overall use and functionality should be unchanged.

g-otn, 2025-03-17:
Thanks, seems to be working fine with patch-package in my project!

Fritzz26, 2025-05-06:
tyyyy

benevolentsocialplanner, 2025-05-26:
"react-native": "0.79.2"
app crashes

saadnaveed94, 2025-06-21:
Your are a lifesaver.

twzhangyang, 2025-06-25:
Thanks, I got a crash and don't know what's happening, it's invaluable.

418442040, 2025-06-25:
react-native": "0.79.4"   
app crashes

sfourdrinier, 2025-07-09:
I am also affected by this problem. 

react-native "0.79.5".

noticed it when I started to have another bluetooth sdk in my app and this at the same time and enabled core-central.

ps73, 2025-07-17:
Crashes in 0.79.5 or with Expo 53.

moonbird-yordi, 2025-09-01:
@mario688  The patch fixed a crash on Expo for me! Thank you 🙏 
Expo v53
React Native v0.79.6
react-native-ble-plx v3.5.0