dotintent/react-native-ble-plx #1279: 🐛 Device not showing up in scan after disconnecting

olalonde, 2025-03-12:
### Prerequisites

- [x] I checked the documentation and FAQ without finding a solution
- [x] I checked to make sure that this issue has not already been filed

### Expected Behavior

I hit a weird bug on Android where after disconnecting from a BLE device, it no longer shows up in the device scan. I have confirmed through "nrf connect for mobile" that the device is indeed advertising and scannable. Sometimes the device will suddenly show up during the scan again. Very strange.

### Current Behavior

It stops showing the device in the scan after disconnecting from it.

### Library version

3.5.0

### Device

Android 15 (pixel 9 pro)

### Environment info

```shell
info Fetching system and libraries information...
System:
  OS: macOS 15.3.1
  CPU: (8) arm64 Apple M1
  Memory: 131.61 MB / 16.00 GB
  Shell:
    version: "5.9"
    path: /bin/zsh
Binaries:
  Node:
    version: 23.9.0
    path: ~/.local/state/fnm_multishells/42435_1741810219145/bin/node
  Yarn: Not Found
  npm:
    version: 10.9.2
    path: ~/.local/state/fnm_multishells/42435_1741810219145/bin/npm
  Watchman:
    version: 2025.03.03.00
    path: /opt/homebrew/bin/watchman
Managers:
  CocoaPods:
    version: 1.16.2
    path: /opt/homebrew/bin/pod
SDKs:
  iOS SDK:
    Platforms:
      - DriverKit 24.2
      - iOS 18.2
      - macOS 15.2
      - tvOS 18.2
      - visionOS 2.2
      - watchOS 11.2
  Android SDK:
    API Levels:
      - "21"
      - "26"
      - "27"
      - "31"
      - "32"
      - "34"
      - "35"
    Build Tools:
      - 30.0.2
      - 30.0.3
      - 31.0.0
      - 34.0.0
[... 278 lines cut ...]
// Create singleton instance
const bleService = new BleService();
export default bleService;
```

### Relevant log output

```shell
(NOBRIDGE) LOG  Starting BLE scan...
 (NOBRIDGE) LOG  Found device: BLE_Relay (E7:15:B1:4E:A4:59)
 (NOBRIDGE) LOG  Found device: BLE_Relay (E7:15:B1:4E:A4:59)
 (NOBRIDGE) LOG  Found device: BLE_Relay (E7:15:B1:4E:A4:59)
 (NOBRIDGE) LOG  Found device: BLE_Relay (E7:15:B1:4E:A4:59)
 (NOBRIDGE) LOG  Connecting to device: E7:15:B1:4E:A4:59
 (NOBRIDGE) LOG  BLE scan stopped
 (NOBRIDGE) LOG  Connected, discovering services and characteristics...
 (NOBRIDGE) LOG  Connected successfully
 (NOBRIDGE) LOG  Disconnected from device
 (NOBRIDGE) ERROR  Notification error: [BleError: Device E7:15:B1:4E:A4:59 was disconnected]
 (NOBRIDGE) LOG  Starting BLE scan...
 (NOBRIDGE) LOG  Found device: M10K3QE (78:9C:85:1B:FF:AF)
 (NOBRIDGE) LOG  Found device: M10K3QE (78:9C:85:1B:FF:AF)
 (NOBRIDGE) LOG  Found device: M10K3QE (78:9C:85:1B:FF:AF)
etc.
```

olalonde, 2025-03-12:
Strangely, when I open the "Nrf Connect for Mobile" app, scan BLE devices and go back to my app, the device suddenly shows up again in my own app. Very strange. Rebooting the app also fixes the issue.

claudioPrisco, 2025-04-01:
same problem

aliberski, 2025-05-26:
@olalonde do you trigger `startDeviceScan` again after disconnecting? It should return a full list of reachable devices

benletchford, 2025-05-29:
@aliberski me and my team also look to be seeing this issue right now

animaonline, 2025-06-11:
Were you able to get this working, @olalonde?

benletchford, 2025-06-18:
@olalonde @claudioPrisco @animaonline @kent-williams @irenmax I may have a potential patch but need assistance validating it properly: #1296 

The way `discoveredDevices` is handled is a little suspect.

kent-williams, 2025-06-18:
@benletchford, thanks for the quick fix attempt!

Unfortunately, I'm not seeing improved behavior yet with the branch. I'm testing with two devices, and I've noticed that the devices eventually show up in the scans if you wait for 30 sec or more.

I'm interested to see what others are seeing.