dotintent/react-native-ble-plx #1295: 🐛 ibeacon not detected

nympheastudio, 2025-06-10:
### Prerequisites

- [x] I checked the documentation and FAQ without finding a solution
- [x] I checked to make sure that this issue has not already been filed

### Expected Behavior

Hello, I purchased an NRF51822 Bluetooth Eddystone iBeacon (specifically a 'Balise/iBeacon/Eddystone' model). I've created a development build using Expo. My issue is that while react-native-ble-plx can detect other BLE devices, it's failing to discover my iBeacon. I've confirmed the iBeacon's presence, name, and MAC address using a separate BLE scanner application.

`{
  "expo": {
    "name": "Helpi",
    "msgMaj": "🚀 Mise à jour de l'application festival OFF",
    "newArchEnabled": false,
    "slug": "helpi",
    "version": "1.0.0",
    "owner": "perodo",
    "android": {
      "versionCode": 100,
      "package": "com....app",
      "googleServicesFile": "./google-services.json",
       "permissions": [
        "android.permission.CAMERA",
        "android.permission.NOTIFICATIONS",
        "android.permission.BLUETOOTH",
        "android.permission.BLUETOOTH_ADMIN",
        "android.permission.BLUETOOTH_CONNECT",
        "android.permission.BLUETOOTH_SCAN",
        "android.permission.ACCESS_FINE_LOCATION",
         "android.permission.ACCESS_COARSE_LOCATION"
      ]
    },
   ...

### Current Behavior

I've verified with another BLE scanner app (showing its name and MAC address), simply doesn't show up in the react-native-ble-plx

### Library version

3.5.0

### Device

redmi note 14

### Environment info

```shell
android 14
```

### Steps to reproduce

1. …nistall     "react-native-ble-plx": "^3.5.0", and  "expo": "^53.0.0",
2. …

### Formatted code sample or link to a repository

```js
import React, { useEffect } from 'react';
import { View, Button, Alert, PermissionsAndroid, Platform } from 'react-native';
import { BleManager, ScanMode } from 'react-native-ble-plx';
import { Buffer } from 'buffer';

const bleManager = new BleManager();

const requestPermissions = async () => {
  if (Platform.OS === 'android') {
    try {
      const granted = await PermissionsAndroid.requestMultiple([
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN,
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT,
        PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
      ]);
      console.log('✅ Permissions:', granted);
    } catch (error) {
      console.error('❌ Permission error:', error);
      Alert.alert('Error', 'Failed to request permissions.');
    }
  }
};

const parseManufacturerData = (data: string | null) => {
  if (!data) return null;
  try {
    const buffer = Buffer.from(data, 'base64');
    const hex = buffer.toString('hex').toUpperCase();
    return hex;
  } catch (err) {
    console.error('Error parsing manufacturerData:', err);
    return null;
  }
};

const isIBeacon = (hex: string | null) => {
  return hex?.startsWith('4C000215'); // Apple iBeacon prefix
};

const BleScanner = () => {
  const startScan = () => {
    console.log('🔍 Starting BLE scan...');
    Alert.alert('Scan', 'Started BLE scan...');

    bleManager.startDeviceScan(
      null,
      { scanMode: ScanMode.LowLatency, allowDuplicates: true },
      (error, device) => {
        if (error) {
          console.error('❌ Scan error:', error);
          Alert.alert('Scan Error', error.message);
          return;
        }

        if (device) {
          const parsed = parseManufacturerData(device.manufacturerData);

          const info = {
            id: device.id,
            name: device.name || '—',
            localName: device.localName || '—',
            rssi: device.rssi,
            manufacturerData: parsed || '—',
          };

          if (isIBeacon(parsed)) {
            console.log('📡 📍 iBeacon detected:', info);
          } else {
            console.log('📡 Other BLE device:', info);
          }
        }
      }
    );
  };

  useEffect(() => {
    requestPermissions();

    return () => {
      console.log('🛑 Stopping BLE scan');
      bleManager.stopDeviceScan();
    };
  }, []);

  return (
    <View style={{ padding: 20 }}>
      <Button title="Start BLE Scan" onPress={startScan} />  
    </View>
  );
};

export default BleScanner;
```

### Relevant log output

```shell
LOG  📡 Other BLE device: {"id": "75:30:F9:95:50:6D", "localName": "—", "manufacturerData": "4C0010063E1E55B5462F", "name": "—", "rssi": -60}..."manufacturerData": "060001092002891A4A8278884F63FA9CD8FE0EDCA3DE2603289729B07B", "name": "—", "rssi": -68}
 LOG  📡 Other BLE device: {"id": "52:06:F2:D7:31:F2", "localName": "—", "manufacturerData": "4C00100631191692A180", "name": "—", "rssi": -74}
 LOG  📡 Other BLE device: {"id": "D0:62:2C:38:13:C5", "localName": "Xiaomi Smart Band 8 13C5", "manufacturerData": "—", "name": "Xiaomi Smart Band 8 13C5", "rssi": -90}
 LOG  📡 Other BLE device: {"id": "75:76:B4:3C:B2:1C", "localName": "—", "manufacturerData": "4C001007361F19C2120438", "name": "—", "rssi": -87}
 LOG  📡 Other BLE device: {"id": "2B:14:08:F3:94:7A", "localName": "—", "manufacturerData": "060001092002293FB53B5E49223496266A7402BFDC78D954DE030FB901", "name": "—", "rssi": -84}
 LOG  📡 Other BLE device: {"id": "46:A1:62:FB:5F:3D", "localName": "—", "manufacturerData": "4C001007391FF5B904CD58", "name": "—", "rssi": -90}
 LOG  📡 Other BLE device: {"id": "08:53:26:2E:1C:06", "localName": "—", "manufacturerData": "060001092022213B6AB796A4283B45D6F151C892B6087F1D998336FDCB", "name": "—", "rssi": -94}
 LOG  📡 Other BLE device: {"id": "15:80:27:CF:51:47", "localName": "—", "manufacturerData": "06000109210AE1FEDCD19C70504F3035383531", "name": "—", "rssi": -91}
 LOG  📡 Other BLE device: {"id": "13:EE:88:A8:05:9C", "localName": "—", "manufacturerData": "—", "name": "—", "rssi": -89}
 LOG  📡 Other BLE device: {"id": "72:78:42:E1:A8:64", "localName": "—", "manufacturerData": "060001092006351D1DA740BD372074538699AB71CAE5F12571958C7A4E", "name": "—", "rssi": -89}
⠙ Stopping server LOG  📡 Other BLE device: {"id": "38:83:51:19:47:2B", "localName": "—", "manufacturerData": "0600010F202297E6597CE181A185C194789AD0A8D5D8DFC32C9A20C15B", "name": "—", "rssi": -78}
 LOG  📡 Other BLE device: {"id": "1C:C8:B2:DA:D9:A8", "localName": "—", "manufacturerData": "060001092022BF6724307EABD290A5413BFC8D6DEB5690813E097F6B22", "name": "—", "rssi": -96}
 LOG  📡 Other BLE device: {"id": "1A:E4:FC:AD:39:67", "localName": "—", "manufacturerData": "060001092002891A4A8278884F63FA9CD8FE0EDCA3DE2603289729B07B", "name": "—", "rssi": -67}
 LOG  📡 Other BLE device: {"id": "00:26:DB:5B:DA:CB", "localName": "—", "manufacturerData": "0600010F202290D313C433F48013876AD3720F10787FC6DD69560D9F0F", "name": "—", "rssi": -94}
⠹ Stopping server LOG  📡 Other BLE device: {"id": "17:73:B7:10:67:B7", "localName": "—", "manufacturerData": "0600010F202280909B5EA20621ED33D7E80653970B21A1EEFB19BA475B", "name": "—", "rssi": -91}
 LOG  📡 Other BLE device: {"id": "2B:14:08:F3:94:7A", "localName": "—", "manufacturerData": "060001092002293FB53B5E49223496266A7402BFDC78D954DE030FB901", "name": "—", "rssi": -80}
 LOG  📡 Other BLE device: {"id": "08:53:26:2E:1C:06", "localName": "—", "manufacturerData": "060001092022213B6AB796A4283B45D6F151C892B6087F1D998336FDCB", "name": "—", "rssi": -99}
 LOG  📡 Other BLE device: {"id": "18:B2:BB:98:40:00", "localName": "—", "manufacturerData": "060001092022C0477E90576BB1BC6E918B77C1020CFB2C8F3FEFDE6A29", "name": "—", "rssi": -96}
› Stopped server
```

necipsunmaz, 2025-06-10:
I have the same problem. Eddystone is not scanning either. I used android device.

nympheastudio, 2025-06-18:
it work now, i had to prebuild my project, then in the androidmanifest   add permission param **tools:remove="android:usesPermissionFlags"**

<manifest xmlns:android="http://schemas.android.com/apk/res/android" xmlns:tools="http://schemas.android.com/tools">
  <uses-permission-sdk-23 android:name="android.permission.ACCESS_COARSE_LOCATION"/>
  <uses-permission-sdk-23 android:name="android.permission.ACCESS_FINE_LOCATION"/>
  <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
  <uses-permission android:name="android.permission.BLUETOOTH"/>
  <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
  <uses-permission android:name="android.permission.BLUETOOTH_CONNECT"/>
  <uses-permission android:name="android.permission.BLUETOOTH_SCAN" tools:remove="android:usesPermissionFlags"/>
  <uses-permission android:name="android.permission.CAMERA"/>
  <uses-permission android:name="android.permission.INTERNET"/>
  <uses-permission android:name="android.permission.NOTIFICATIONS"/>
  <uses-permission android:name="android.permission.READ_CALENDAR"/>
  <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE"/>
  <uses-permission android:name="android.permission.RECORD_AUDIO" tools:node="remove"/>
  <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW"/>
  <uses-permission android:name="android.permission.VIBRATE"/>
  <uses-permission android:name="android.permission.WRITE_CALENDAR"/>
  <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE"/>
  <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>...  

now the ibeacon is founded : 

Android Bundled 5722ms node_modules/expo/AppEntry.js (1371 modules)
 LOG  ✅ Permissions accordées
 LOG  🔍 Démarrage du scan BLE…
 LOG  📡 Autre device → 64:2F:CD:9D:32:4C RSSI=-98 mfg=4C0010050218C8DF86
 LOG  📍 iBeacon détecté → F2:75:55:F9:FE:E0 RSSI=-57 mfg=4C000215FDA50693A4E24FB1AFCFC6EB07647825271B4CB9C9